openwrt/packages #3650: mosquitto: the version 1.4.10 have a fatal compile problem need to be update

KerwinKoo, 2016-12-12:
Maintainer: @eclipse
Environment: LEDE master b305b8c38692e2345a6c580baf0b1eda929ce5d3

Description:

There is a fatal bug when compile the mosquitto installed by `script/feeds install` :

```
websockets.o: In function `callback_mqtt':
websockets.c:(.text+0x104): undefined reference to `libwebsocket_get_protocol'
```

this bug has been fixed by offical `https://github.com/eclipse/mosquitto` in issue [1.4.10 compiling libwebsockets error #272](https://github.com/eclipse/mosquitto/issues/272). BUT the version in LEDE haven't updated to the latest version.

karlp, 2016-12-12:
how are you building?  because http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/mosquitto_1.4.10-1_mipsel_24kc.ipk is clearly being built still, along with http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/libwebsockets-full_2.0.3-1_mipsel_24kc.ipk

You would only be seeing that error if you're trying to compile against an older libwebsockets? And the 15.05 branch already contains the patch: https://github.com/openwrt/packages/blob/for-15.05/net/mosquitto/patches/0001-websockets-fix-compatibility-with-older-lws-versions.patch  (I was the one who reported it upstream myself, so I'm rather surprised that you claim to be hitting this)

closed 2016-12-12 by karlp

KerwinKoo, 2016-12-13:
@karlp thanks very much. 

This patch is very useful and important for me but I didn't get it automatically by `./script/feeds install -a -p mosquitto`. Is package's patches need created by user?