openwrt/packages #3657: jool: PR 3604 breaks ar71xx build

anomeome, 2016-12-15:
[This](https://github.com/openwrt/packages/pull/3604) applied to current LEDE build yields [this](http://ix.io/1MxL) in my source tree, building to target C7V2 C5V1

hnyman, 2016-12-16:
compiling jool for ar71xx breaks also at buildbot:
http://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/jool/compile.txt

but compiles ok for e.g. x86_64 and arm_cortex-a15_neon-vfpv4

cc to @danrl and @jow-

danrl, 2016-12-16:
I asked at the upstream mailing list for their point of view. I am not deep enough into the code to understand why it fails on MIPS and succeeds on X86. Will take some time.

jow-, 2016-12-16:
Upstream likely needs a change like https://patchwork.kernel.org/patch/3233311/

ydahhrk, 2016-12-16:
> Upstream likely needs a change like https://patchwork.kernel.org/patch/3233311/

Agree. It crashed very early, though. Did you try adding `#include net/ip6_checksum.h` to `Jool/mod/stateful/bib/db.c`, just to make sure it's all that's going to be needed?

danrl, 2016-12-16:
I'll try that later today and let you know.

danrl, 2016-12-16:
Patch to upstream is underway. Will bump package version once it is accepted or otherwise fixed in upstream. Thanks for the reports!

commit 0510753c (2016-12-17): jool: version bump (fixes #3657)

Latest upstream changes explicitly include `net/ip6_checksum.h`.
Fixes issue #3657.
closed 2016-12-17 by yousong