openwrt/packages #3667: Asterisk: Some Ideas and suggestions?

LipkeGu, 2016-12-17:
Currently im needing asterisk whe i started to configure the highly complicated & and overblown Config (which needs alot of space on the flash) I started to strip the config down to a bare metal part for SIP and LEDE (Fritzbox 7360SL). Starting with a UCI config it allows me to remove some files and makes it able to configure a basic PBX.

I started to create a UCI config file:
```
config asterisk
	option verbose				'0'
	option dontwarn			'yes'
	option maxcalls				'10'
	option maxload				'0.9'
	option minmemfree			'1'
	option tansmit_silence		'yes'
	option language			'en'
	option systemname			'LEDE'
	option autosystemname		'yes'
	option mindtmfduration		'80'
# user
	option userbase			'6000'
# sip
	option port				'5060'
	option bindaddr			'0.0.0.0'
	option context				'default'
	option srvlookup			'yes'
# rtp
	option rtpstart				'10000'
	option rtpend				'20000'

config asterisk_user
	option fullname			'Joe User'
	option description			'Sipura SPA 921'
	option email				'joe@foo.bar'
	option secret				'1234'
	option vmsecret			'1234'
	option hasvoicemail			'no'
	option hassip				'yes'
	option hasiax				'no'
	option hash323			'no'
	option hasmanager			'no'
	option callwaiting			'no'
	option macaddress			'112233445566'
	option autoprov			'yes'
	option linekeys				'1'
	option linenumber			'1'
	option cidnumber			'6000'
	option treewaycalling		'yes'
	option callwaitingcallerid	'yes'
	option transfer				'yes'
	option canpark				'yes'
	option cancallforward		'yes'
	option callreturn			'yes'
	option callgroup			'1'
	option pickupgroup			'1'
	option nat				'no'

config asterisk_register
	option username			'12345678e0'
	option password			'SIPPASS'
	option domain				'sipgate.de'

config asterisk_peer
	option number				'6000'
	option type				'friend'
	option secret				'1234'
	option host				'dynamic'
	option nat				'yes'
	option qualify				'yes'

config asterisk_peer
	option number				'12345678e0'
	option type				'peer'
	option secret				'SIPPASS'
[... 340 lines cut ...]
exten => _[0-1].,8,Hangup()

[sipgate-in]
exten => 12345678e0,1,Answer()
exten => 12345678e0,2,Wait(2)
exten => 12345678e0,3,GotoIf($["${CALLERID(num)}" = "Unknown"]?unknown_cid)
exten => 12345678,4,GotoIf($["${CALLERID(num)}" = "Unbekannt"]?unknown_cid)
exten => 12345678e0,5,GotoIf($["${CALLERID(num)}" = "Privat"]?unknown_cid)
exten => 12345678e0,6,GotoIf($["${CALLERID(num)}" = "Private"]?unknown_cid)
exten => 12345678e0,7,GotoIf($["${CALLERID(num)}" = "0000000000"]?unknown_cid)
exten => 12345678e0,8,Playback(pls-hold-while-try&one-moment-please)
exten => 12345678e0,9,Wait(2)
exten => 12345678e0,10,Dial(SIP/6000,20,r)
exten => 12345678e0,11,Wait(2)
exten => 12345678e0,12,Playback(number-not-answering&please-try-call-later,noanswer)
exten => 12345678e0,13,Hangup()
exten => 12345678e0,14(unknown_cid),Playback(sorry&privacy-unident&goodbye)
exten => 12345678e0,15,Hangup()

[default]
include => sipgate-out
include => my phones
```

Can anyone post me a snipped for the uci config? It would be nice when we can put the extensions.conf in a uci config.

LipkeGu, 2016-12-17:
Ok i looked in the example of the timeservers and found a solution... and it works :D

```
config asterisk_dialplan
        option context                  'my_phones'
        option pattern                  '_[2-9]000'
        option source                   'exten'
        list entries                    '1,Answer()'
        list entries                    '2,Wait(2)'
        list entries                    '3,Playback(pls-hold-while-try&one-moment-please)'
        list entries                    '4,Wait(1)'
        list entries                    '5,Dial(SIP/${EXTEN},20)'
        list entries                    '6,Playback(number-not-answering&please-try-call-later,noanswer)'
        list entries                    '7,(unknown_cid),Playback(sorry&privacy-unident&goodbye)'
        list entries                    '8,Hangup()'

config asterisk_dialplan
        option context                  'sipgate-out'
        option pattern                  '_[0-1].'
        option source                   'exten'
        list entries                    '1,Answer()'
        list entries                    '2,Wait(2)'
        list entries                    '3,Playback(pls-hold-while-try&one-moment-please)'
        list entries                    '4,Wait(1)'
        list entries                    '5,Dial(SIP/${EXTEN}@xxxxxxxxe0,40,r)'
        list entries                    '6,Playback(number-not-answering&please-try-call-later,noanswer)'
        list entries                    '7(unknown_cid),Playback(sorry&privacy-unident&goodbye)'
        list entries                    '8,Hangup()'

config asterisk_dialplan
        option context                  'sipgate-in'
        option pattern                  'xxxxxxxxe0'
        option source                   'exten'
        list entries                    '1,Answer()'
        list entries                    '2,Wait(2)'
        list entries                    '3,Playback(pls-hold-while-try&one-moment-please)'
        list entries                    '4,Wait(2)'
        list entries                    '5,Dial(SIP/6000,20,r)'
        list entries                    '6,Playback(number-not-answering&please-try-call-later,noanswer)'
        list entries                    '7,Hangup()'

config asterisk_dialplan
        option context                  'default'
        list includes                   'sipgate-out'
        list includes                   'my_phones'
```

hnyman, 2016-12-31:
I close this because Asterisk lives in https://github.com/openwrt/telephony

closed 2016-12-31 by hnyman