openwrt/packages #3670: bind: Set --enable-filter-aaaa to be on by default?

jamesmacwhite, 2016-12-18:
Maintainer: **@nmeyerhans**

Description:

I use the bind package in Entware-ng, which uses OpenWRT packages as the upstream source. Would you be against having the `--enable-filter-aaaa` flag on by default? The reason I ask is because currently anyone wanting this functionality must compile from source with the option explicitly enabled. It would be nice if Entware and other sources built the package with it on by default.

https://github.com/Entware-ng/Entware-ng/issues/315

I thought it would be best to report upstream to see if this was possible/discuss.

jamesmacwhite, 2017-01-07:
@nmeyerhans Just wondering if you've managed to have a look at this at all?

nmeyerhans, 2017-01-08:
Yeah, this is probably fine. I can't imagine that this will add significantly the the binary size, which would be my main concern. I haven't actually checked that, though...

jamesmacwhite, 2017-01-08:
@nmeyerhans 

Thanks for your reply.

I don't believe it does, if there is an increase its only a few kilobytes. However if it would help, I can compile with the flag set against the latest version of bind and compare the ipk sizes for validation?

nmeyerhans, 2017-01-08:
Don't bother; I've already done it. Looks like it adds all of 72 bytes to the armv7l `named` binary.

I'll merge the change shortly...

nmeyerhans, 2017-01-09:
Pushed 909209e7531de2ea63f1f298adce985406d8ba08 to master.

closed 2017-01-09 by nmeyerhans