openwrt/packages #3678: ddns-scripts: build error (certificate problems)

juanriccio, 2016-12-19:
for FILE in `find /lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/files -type f`; do /lede-source/staging_dir/host/bin/sed -i -e 's/^\s*#/#/' -e '/^#\s\|^#$/d' -e 's/\s#\s.*$//' -e 's/\s*$//' -e '/^\/\/\s/d' -e '/^\s*$/d'	$FILE; done
# ensure that VERSION inside dynamic_dns_functions.sh reflect PKG_VERSION of Makefile
/lede-source/staging_dir/host/bin/sed -i -e '/^VERSION=*/s/.*/VERSION="2.7.6-6"/' /lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/files/dynamic_dns_functions.sh
cp -fpR ./tools /lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6
chmod 755 /lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/tools/*
/lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/tools/public_suffix_list.sh
--2016-12-19 12:10:03--  https://publicsuffix.org/list/public_suffix_list.dat
Resolving publicsuffix.org... 63.245.213.24
Connecting to publicsuffix.org|63.245.213.24|:443... connected.
ERROR: certificate common name `static-san.mozilla.org' doesn't match requested host name `publicsuffix.org'.
To connect to publicsuffix.org insecurely, use `--no-check-certificate'.
make[3]: *** [/lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/.built] Error 1
make[3]: Leaving directory `/lede-source/feeds/packages/net/ddns-scripts'
make[2]: *** [package/feeds/packages/ddns-scripts/compile] Error 2
make[2]: Leaving directory `/lede-source'
make[1]: *** [/lede-source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory `/lede-source'
make: *** [world] Error 2

chris5560, 2016-12-19:
When did you last update certificates on you build machine ?
Try "wget -O- https://publicsuffix.org/list/public_suffix_list.dat" from console
I checked build on XUbuntu 16.10. Installed ca-certificates version "20160104ubuntu1"

juanriccio, 2016-12-20:
My last update was a long time ago, since my build machine runs a very old Ubuntu distribution, not supported anymore. I can't upgrade my Ubuntu on a newer version - not on that hardware anyway, until I get myself a new, decent machine.

I understand it's a bit off topic, but if you have useful info could you share any pointer on manually installing the new certificate(s) that are needed? I tried several commands, incuding
`sudo apt-get install ca-certificates` ,
finally resulting in
```ca-certificates is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.```

Now it seems current:
```$ ls -l /etc/ssl/certs/cacert.pem 
-rw-r--r-- 1 root root 263596 2016-11-02 05:12 /etc/ssl/certs/cacert.pem
```
but still that package won't build.

chris5560, 2016-12-20:
Do you really need to build ddns-scripts for your box ?
You can easily download from openwrt package servers because it's a CPU independent package.
Other option is to copy the needed files to the correct locations.
If you not using CloudFlare's API-Version 1 as DDNS provider you do not need public_suffix_list.dat.

dedeckeh, 2016-12-23:
@chris5560 The introduction of the script public_suffix_list.dat in commit https://github.com/openwrt/packages/commit/1c20dcb71a693892a72882c6594c66a05239cea0#diff-8fcfb85f765ff1326b37d7791feb67bc not only creates compile issues on some older distributions but more importantly makes it impossible to create reproducible builds.
This has severe drawbacks which has been discussed by Holger Levsen in his talk about reproducible builds on the OpenWRT summit 2016 (http://openwrtsummit.org/slides/reproducible.pdf).
Would it be possible to avoid the download of the public_suffix_list.dat in the compile step of ddns as ddns is a quite popular package due the numerous supported dyndns providers and its great flexibility

chris5560, 2016-12-24:
@dedeckeh
I replaced https: with http: request to read public_suffix_list.dat to fix certificate problems on older building machines.

The problems with public_suffix_list.dat:
- lots of changes during the year
- not really a release management
- only used for CloudFlare.com v1 updating
so during build you always get the latest version.

Long term it will no longer be needed if users have migrated to V4 protokoll.
I will try to find a way to migrate existing Cloudflare_V1 settings inside ddns config file to no longer need public_suffix_list.dat at runtime.
Because I currently see no way to inform users after package upgrade that they need to modify their config files, it's still needed during package upgrade.

xdarklight, 2016-12-24:
@chris5560 I think the problem is not the fact that ```public_suffix_list.dat``` is needed but rather the way it's being downloaded.
Can't we use their github repo for accessing specific versions of that file (at least when creating the ddns-scripts package)? example link: https://github.com/publicsuffix/list/blob/6911a4ab68fb5fe47f24c6e7690b67a03a54c1bf/public_suffix_list.dat

I also had a failing LEDE build because I had no internet connection for a few minutes, which resulted in a build failure due to:
```
...
failed: Name or service not known.
wget: unable to resolve host address 'publicsuffix.org'
make[3]: *** [Makefile:366: /mnt/devel/lede/stable/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/.built] Error 1
make[3]: Leaving directory '/mnt/devel/lede/stable/feeds/packages/net/ddns-scripts'
...
```

juanriccio, 2016-12-25:
It doesn't build yet:
```
/lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/tools/public_suffix_list.sh
--2016-12-25 12:12:09--  http://publicsuffix.org/list/public_suffix_list.dat
Resolving publicsuffix.org... 63.245.213.24
Connecting to publicsuffix.org|63.245.213.24|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://publicsuffix.org/list/public_suffix_list.dat [following]
--2016-12-25 12:12:10--  https://publicsuffix.org/list/public_suffix_list.dat
Connecting to publicsuffix.org|63.245.213.24|:443... connected.
ERROR: certificate common name `static-san.mozilla.org' doesn't match requested host name `publicsuffix.org'.
To connect to publicsuffix.org insecurely, use `--no-check-certificate'.
make[3]: *** [/lede-source/build_dir/target-mips_24kc_musl-1.1.15/ddns-scripts-2.7.6/.built] Error 1
make[3]: Leaving directory `/lede-source/feeds/packages/net/ddns-scripts'
make[2]: *** [package/feeds/packages/ddns-scripts/compile] Error 2
make[2]: Leaving directory `/lede-source'
make[1]: *** [/lede-source/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_compile] Error 2
make[1]: Leaving directory `/lede-source'
make: *** [world] Error 2

```

hnyman, 2016-12-31:
Should be fixed with #3701

closed 2016-12-31 by hnyman