openwrt/packages #3695: canutils Build Fail

riodda, 2016-12-24:
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (x86_64, APU2, LEDE )

Description:
Despite haveing selected canutils canutils subpacakges fail to compile due to failed depedencies
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-log-conversion:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-log-conversion.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-canbusload:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-canbusload.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-can-calc-bit-timing:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-can-calc-bit-timing.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-candump:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-candump.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-cangen:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-cangen.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-cangw:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-cangw.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-canplayer:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-canplayer.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-cansniffer:
 * 	canutils * 
 * opkg_install_cmd: Cannot install package canutils-cansniffer.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for canutils-cansend:
 * 	canutils * 
```

riodda, 2016-12-24:
Commenting the dependecies in the Makefile the errors are gone but the build lacks several of the canutils execs such as slcand slcan_attach

yegorich, 2017-03-13:
@riodda have you already looked at the patch?

riodda, 2017-03-14:
Nope, i will

closed 2017-03-15 by hnyman