openwrt/packages #3699: rp-pppoe: Packaging wish: include pppoe client executable

arnysch, 2016-12-24:
Environment:   rp-pppoe package update  f6611ab on 23 Oct

For DSL testing and checking, I find the the user space client program '**pppoe**' most helpful ("pppoe -A").

Though pppoe is built in rp-pppoe, it is not packaged for installation.

I added a package for it to my local rp-pppoe/Makefile manually  (URL below as an example).
It would be nice if it could come by default in OpenWrt/LEDE.

Just some ideas:
- Have package help explain that package is present for testing only, and that ppp-mod-pppoe is preferred for normal operation?
- Call the package 'rp-pppoe-client'?
- Rename executable 'pppoe' to 'pppoe-client'?
- Let package not depend on package ppp, because it may be used for testing without ppp?
- Let package depend on @broken, so it is not offered by default?

For example:   http://arny.tjps.eu/OpenWrt/misc/20161224/Makefile

cshoredaniel, 2017-01-12:
@arnysch Would you be interested in taking over maintainership?  I don't have a need for this package any more, but don't want to orphan it for those that do so have been in 'maintenance mode' with this package.

arnysch, 2017-01-12:
Hi Daniel,

due to little time I shy away from becoming a maintainer.
Because this means taking over responsibility :-)
Just once in a while I mess about with OpenWrt for my little home router.
I do not know the maintainer procedures to heed.

Also, from rp-pppoe I only need the client pppoe, just for testing
connectivity above the level of DSL sync.

Testing all rp-pppoe functionality would be too much for me,
so I could only package and wait for other people suggestions about what
to modify.

Having said this:
- If you want to get rid of rp-pppoe maintenance
- If you think that rp-pppoe is a less critical packages in OpenWrt/Lede
   because only few people use it
- If you think rp-pppoe is a good starting point to become a maintainer
I am willing to try. Some hints (links or short remarks) about
procedures/what to watch out for would then be helpful.

Regards,
Arnold

cshoredaniel, 2017-01-12:
Would you be more comfortable with doing a co-maintainer thing? You could do this patch, and help with DSL testing (I don't have DSL; I was using this to test DSL-type pppoe when I was doing devel that I waned to make sure work for more use cases than just static and DHCP.  I think this package is not a critical package because it's not widely used.  It should be mostly a case of doing version updates and dealing with compile issues due to trunk changes (or changes in the new versions).

arnysch, 2017-12-10:
**Just discovered this existing alternative:**

CONFIG_PACKAGE_pppoe-discovery:

This tool performs the same discovery process as pppoe, but does
not initiate a session. Can be useful to debug pppoe.
https://ppp.samba.org/
Felix Fietkau <nbd@nbd.name>

Symbol: PACKAGE_pppoe-discovery [=y]
Type  : tristate
Prompt: pppoe-discovery........................ Perform a PPPoE-discovery process
Location:
     -> Network│  
       -> ppp-multilink........................ PPP daemon (with multilink support) (PACKAGE_ppp-multilink [=n])

closed 2017-12-10 by arnysch