openwrt/packages #3706: <libustream-polarssl>: can't build lede because of file clashes

finasfinas, 2016-12-26:
this is the error:

Description:
Configuring bcg729.
Configuring dnsmasq-full.
Collected errors:
 * check_data_file_clashes: Package libustream-polarssl wants to install file /home/finas/lede.C5/build_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package libustream-polarssl.
 * check_data_file_clashes: Package libustream-polarssl wants to install file /home/finas/lede.C5/build_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package cshark.
package/Makefile:60: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/finas/lede.C5'
package/Makefile:130: recipe for target '/home/finas/lede.C5/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_install' failed
make[1]: *** [/home/finas/lede.C5/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/finas/lede.C5'
/home/finas/lede.C5/include/toplevel.mk:197: recipe for target 'world' failed
make: *** [world] Error 2

diizzyy, 2016-12-30:
@finasfinas 
It's fixed in this pull request https://github.com/lede-project/source/pull/655

finasfinas, 2017-01-02:
still giving me the same error after make dirclean

Configuring dnsmasq-full.
Collected errors:
 * check_data_file_clashes: Package libustream-polarssl wants to install file /home/finas/lede.C5/build_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package libustream-polarssl.
 * check_data_file_clashes: Package libustream-polarssl wants to install file /home/finas/lede.C5/build_dir/target-mips_24kc_musl-1.1.15/root-ar71xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package cshark.
package/Makefile:60: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 255
make[2]: Leaving directory '/home/finas/lede.C5'
package/Makefile:130: recipe for target '/home/finas/lede.C5/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_install' failed
make[1]: *** [/home/finas/lede.C5/staging_dir/target-mips_24kc_musl-1.1.15/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/finas/lede.C5'
/home/finas/lede.C5/include/toplevel.mk:197: recipe for target 'world' failed

finasfinas, 2017-01-02:
and of course after git pull feeds update -a and install -a

hnyman, 2017-01-02:
The underlying issue is that polarssl SSL library is EOL and is being phased out.

Luci-ssl was changed to use ustreamssl-mbedtls (instead of -polarssl) a few days ago and that triggered your trouble, as there can be just one libustream variant in use. And in your case cshark tries to use libustream-polarssl.

> opkg_install_cmd: Cannot install package cshark.

cshark is one of the few packages that still uses libustream-polarssl. There is a PR https://github.com/openwrt/packages/pull/3730 switching cshark to use mbedtls instead, but the cshark maintainer @lperkov has not yet approved the PR.

If you want to fix your build, you can edit the cshark Makefile  in your feed repo ( feeds/packages/net/cshark/Makefile ) and make one change. On the DEPENDS line, change the last item to be `+libustream-mbedtls` instead of `+libustream-polarssl`. See below:
```
-  DEPENDS:=+libjson-c +libpcap +libuci +libubox +libuclient +libustream-polarssl
+  DEPENDS:=+libjson-c +libpcap +libuci +libubox +libuclient +libustream-mbedtls
```

hnyman, 2017-01-02:
cshark PR #3730 has been merged. You should be able to compile cshark again.

But check first that your .config (e.g. via menuconfig) does not have libustream-polarssl selected. Deselect it manually, if needed.

Please close the issue if it works again.

finasfinas, 2017-01-02:
working perfectly now. thanks!

closed 2017-01-02 by finasfinas