openwrt/packages #3709: [Request sound packages] Spotify Connect

schmurtzm, 2016-12-27:
Hi, for spotify users, "spotify connect" is really great because it works like a remote : your smartphone don't read the stream and send it to the audio device, it ask to the device directly to read a stream from spotify so you can switch off your smartphone it continue to play the playlist.

Recently I've buyed a linux audio device with spotify connect available on it. In sources I find some traces of github repo.

So now it is possible to have spotify connect on raspberry (with volumio for example) and some Linux boards have successfully implement it with the help of code sources below.

So I think this is possible to implement it in openWRT, just missing a competent developer to do it  ;)

There are 3 repos about spotify connect on github :
https://github.com/sashahilton00/spotify-connect-resources
https://github.com/Fornoth/spotify-connect-web
https://github.com/badfortrains/spotcontrol

How to install on raspberry here :
https://discourse.osmc.tv/t/howto-setup-a-spotify-connect-web-server-on-a-raspberry-pi-with-osmc/15818

It's not very complex.

May be with the same principle you can make it work on openwrt ?

Thanks to this great community !

hnyman, 2016-12-31:
Feel free to submit a pull request creating the package here.

closed 2016-12-31 by hnyman

schmurtzm, 2017-01-04:
I'm not able to create it by myself... 
Can I create a "pull request" to ask to the community to create it ?

thess, 2017-01-04:
@schmurtzm - The best place to make this request would be the LEDE and/or OpenWrt forums. Perhaps you can find an interested party there to take ownership, etc.

In order to create a "pull-request" you need to supply the package Makefile and any additional files or patches. You need to have tested your submission and you must sign it. Please read and follow our guide for submissions found here:
https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md