openwrt/packages #3710: gpsd: Build fails due to missing -lncurses flag

deltahotel, 2016-12-27:
Maintainer: @psidhu 
Environment: LEDE lede-project/source@d3489d899a, packages c785ad3, mips-24k

Description:

`make package/gpsd/compile` fails with due to unresolved symbols:

cgps.o: In function \`die':
cgps.c:(.text+0x2): undefined reference to \`isendwin'
cgps.c:(.text+0x7c): undefined reference to \`mvcur'
cgps.c:(.text+0x82): undefined reference to \`echo'
cgps.c:(.text+0x88): undefined reference to \`endwin'
cgps.c:(.text+0xa8): undefined reference to \`LINES'
cgps.c:(.text+0xac): undefined reference to \`COLS'
…

Both `gpsd` as well as `libncurses` were selected as `m` (module) and `libncurses` is indeed available. Adding an `-lncurses` flag to gcc's options resolves the problem so it's seems it's just this flag which is missing.

RussellSenior, 2017-01-20:
@deltahotel I am seeing this on an up-to-date ArchLinux build host, but not on a fresh checkout on Ubuntu 16.04.  The problem seems to be in the scons detection of the ncurses library.  On Ubuntu the cgps compile command line ends in -lncurses -ltinfo, but these are missing on the ArchLinux build host.

RussellSenior, 2017-01-20:
@deltahotel @psidhu #3888 should fix it.

See: https://github.com/openwrt/packages/pull/3888

deltahotel, 2017-01-22:
Sounds plausible since I was using Arch as well. Thanks for the research and the patch, @RussellSenior! I'll close this issue once your PR is merged.

psidhu, 2017-01-23:
@deltahotel Please test.

glycoknob, 2017-02-19:
Works fine on current Arch, could this be backported to the 17.01 branch?

deltahotel, 2017-03-29:
@psidhu Compilation works fine for me, thanks for the fix! I'll leave the issue open for the moment since @glycoknob's comment is still unanswered.

glycoknob, 2017-05-08:
The issue is still open - there is already a pull request waiting for almost two weeks: https://github.com/openwrt/packages/pull/4300 - is there any problem merging this or is everyone busy / overworked? This breaks building an image with olsr for for all ncurses6 distros at the moment.

hnyman, 2017-05-08:
#4300 was merged. Closing this.

closed 2017-05-08 by hnyman