openwrt/packages #3715: openvswitch: Unable to install openvswitch on 15.05.1

dromie, 2016-12-30:
Maintainer: @commodo Alexandru Ardelean <ardeleanalex@gmail.com>
Environment: ar71xx, NetGear WNDR3700v4, OpenWRT 15.05.1

Error message:
```
root@OpenWrt:~# opkg install openvswitch
Installing openvswitch (2.3.90-58be9c9fd732b5bdd3d4c2e9b8cc2313f570094d) to root...
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/nand/packages/packages/openvswitch_2.3.90-58be9c9fd732b5bdd3d4c2e9b8cc2313f570094d_ar71xx.ipk.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for openvswitch:
 * 	kernel (= 3.18.21-1-c7b319e4772fd435398c9177fe850dff) * 
 * opkg_install_cmd: Cannot install package openvswitch.

root@OpenWrt:~# uname -a
Linux OpenWrt 3.18.23 #1 Sun Jan 31 18:23:08 CET 2016 mips GNU/Linux

```

dromie, 2016-12-30:
Recompiling openvswitch 2.3.90 package using OpenWRT-SDK-15.05.1 and git hash 3c78eb04b8808e869313be3daafc5537e7e7ae43 (packages repo) seems to work.

hnyman, 2016-12-30:
The 15.05.1 release was partially broken. It contains mixed 3.18.21 / 3.18.23 binaries at least for ar71xx. 

The issue has been discussed in:
https://dev.openwrt.org/ticket/22073
https://dev.openwrt.org/ticket/22212
> There are 5 kernel modules for 3.18.21 in ar71xx 15.05.1 packages feed download directory. It looks like they have been compiled earlier, and have not been refreshed when the 15.05.1 release build with kernel 3.18.23 has been built. 

Recompiling packages by yourself using the SDK is likely the best fix (as you already noticed). Alternatively, recompile the whole firmware from the head of the 15.05 branch including the package that has dependency problems.

I close the issue here, as it is not directly related to the openvswitch package itself (in this repo), but to a broken Openwrt release compilation process in 2015, and there are several bugs open in the old Openwrt bug tracker about this.

closed 2016-12-30 by hnyman