openwrt/packages #3722: transmission-cli-openssl and transmission-daemon-openssl installed libpolarssl as dependency

lando-s, 2016-12-31:
-------------------------------

Maintainer: @obsy 
Environment: mips_24kc ar71xx tl-wr1043nd-v2 LEDE-r2709-b7677f0

Description:
using imagebuilder-ar71xx:
package transmission-cli-openssl and/or transmission-daemon-openssl installed libpolarssl as dependency.

```
lede-imagebuilder-ar71xx-generic.Linux-x86_64$ make clean
lede-imagebuilder-ar71xx-generic.Linux-x86_64$ make image PROFILE="DEVICE_tl-wr1043nd-v2" PACKAGES="transmission-cli-openssl" | grep libpolarssl
Installing libpolarssl (1.3.18-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/mips_24kc/base/libpolarssl_1.3.18-1_mips_24kc.ipk.
Configuring libpolarssl.
```
```
lede-imagebuilder-ar71xx-generic.Linux-x86_64$ make clean
lede-imagebuilder-ar71xx-generic.Linux-x86_64$ make image PROFILE="DEVICE_tl-wr1043nd-v2" PACKAGES="transmission-daemon-openssl" | grep libpolarssl
Installing libpolarssl (1.3.18-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/mips_24kc/base/libpolarssl_1.3.18-1_mips_24kc.ipk.
Configuring libpolarssl.
```

please also change ssl dependency from polarssl to mbedtls

obsy, 2016-12-31:
transmission-cli-openssl/transmission-daemon-openssl not depends on libpolarssl (check in control file).

But libcurl has. Use libcurl with other dependency.

obsy, 2016-12-31:
About polarssl dependency see also https://github.com/lede-project/source/pull/655

hnyman, 2016-12-31:
Like @obsy says, the dependency comes via libcurl that currently defaults to polarssl. 
Curl will likely be switched at some point to default into libmbedtls.

If you build the firmware from sources, you can select which SSL library libcurl will use (e.g. openssl). But the compiled packages available in the release repo will use the current default SSL.

I close this as there is not much that be done in tranamission itself (except that @obsy should eliminate the polarssl variant or replace it with one based on mbedtls.)

closed 2016-12-31 by hnyman