openwrt/packages #3728: Unbound: resolving router returns localhost (127.0.0.1)

Borromini, 2016-12-31:
Maintainer: @EricLuehrsen 
Environment: ramips/mt7621, LEDE r2709+1

Description:

Hi Eric, hope this isn't another PEBKAC again ;) When I ping my router from a client, instead of returning its IP, it returns 127.0.0.1:

```
$ ping -c 5 zeus
PING zeus.lan (127.0.0.1) 56(84) bytes of data.
64 bytes from localhost (127.0.0.1): icmp_seq=1 ttl=64 time=0.034 ms
64 bytes from localhost (127.0.0.1): icmp_seq=2 ttl=64 time=0.043 ms
64 bytes from localhost (127.0.0.1): icmp_seq=3 ttl=64 time=0.045 ms
64 bytes from localhost (127.0.0.1): icmp_seq=4 ttl=64 time=0.040 ms
64 bytes from localhost (127.0.0.1): icmp_seq=5 ttl=64 time=0.045 ms

--- zeus.lan ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4057ms
rtt min/avg/max/mdev = 0.034/0.041/0.045/0.007 ms
```

In my browser, surfing to http://zeus does not work. Other hosts on the network return their respective IP (and their web servers are reachable).

My unbound settings:

```
# uci show unbound
unbound.@unbound[0]=unbound
unbound.@unbound[0].dnsmasq_gate_name='0'
unbound.@unbound[0].dnsmasq_link_dns='1'
unbound.@unbound[0].dnsmasq_only_local='0'
unbound.@unbound[0].edns_size='1280'
unbound.@unbound[0].listen_port='53'
unbound.@unbound[0].localservice='1'
unbound.@unbound[0].manual_conf='0'
unbound.@unbound[0].query_minimize='0'
unbound.@unbound[0].rebind_localhost='0'
unbound.@unbound[0].rebind_protection='1'
unbound.@unbound[0].recursion='default'
unbound.@unbound[0].resource='medium'
unbound.@unbound[0].root_age='28'
unbound.@unbound[0].ttl_min='120'
unbound.@unbound[0].unbound_control='0'
unbound.@unbound[0].validator='0'
unbound.@unbound[0].validator_ntp='1'
```

Localhost on the client:
```
$ cat /etc/hosts
127.0.0.1	localhost
127.0.1.1	okeanos.lan	okeanos
```

Thanks and a happy new year

EricLuehrsen, 2016-12-31:
Could you `# uci show dhcp` (redact PII if need be, or email me personally)? We need dnsmasq to be on a different `port` than Unbound (1053 or something). We need dnsmasq init scripts to label the router as a virtual DHCP client `add_local_host`. As a test of dnsmasq alone, you can try this locally at dnsmasq port to bypass Unbound `# drill zeus.lan @127.0.0.1:1053`. It seems Unbound is forwarding `*.lan. IN A` queries for everything, so it must be something weird.

EricLuehrsen, 2016-12-31:
Could you `# uci show dhcp | grep dnsmasq` (redact PII if need be)? We are looking for `port` (1053 or something), option `noresolv` and `nohosts` to prevent dnsmasq from poisoning its own response,  and option `add_local_host` for dnsmasq to label the local host as virtual DHCP client. As a test of dnsmasq alone, you can try this locally at dnsmasq port to bypass Unbound `# drill zeus.lan @127.0.0.1:1053`. It seems Unbound is forwarding `*.lan. IN A` queries for everything, so it must be something weird.

```
config dnsmasq
	option add_local_host '1'
	option authoritative '1'
	option boguspriv '1'
	option domain 'lan'
	option domainneeded '1'
	option expandhosts '1'
	option filterwin2k '0'
	option leasefile '/tmp/dhcp.leases'
	option local '/lan/'
	option localise_queries '1'
	option localservice '1'
	option nohosts '1'
	option nonegcache '1'
	option noresolv '1'
	option port '1053'
	option readethers '0'
	option rebind_localhost '0'
	option rebind_protection '1'
```

EricLuehrsen, 2017-01-01:
Could you `# uci show dhcp | grep dnsmasq` (redact PII if need be)? We are looking for `port` (1053 or something), option `noresolv` (avoid /tmp/resolv.conf) and `nohosts` (avoid /etc/hosts) to prevent dnsmasq from poisoning its own response,  and option `add_local_hostname` for dnsmasq to label the local host as virtual DHCP client. As a test of dnsmasq alone, you can try this locally at dnsmasq port to bypass Unbound `# drill zeus.lan @127.0.0.1:1053`. It seems Unbound is forwarding `*.lan. IN A` queries for everything, so it must be something weird.  _guess: dnsmasq is pulling the host self named loop back entry from /etc/hosts._

```
config dnsmasq
	option add_local_hostname '1'
	option authoritative '1'
	option boguspriv '1'
	option domain 'lan'
	option domainneeded '1'
	option expandhosts '1'
	option filterwin2k '0'
	option leasefile '/tmp/dhcp.leases'
	option local '/lan/'
	option localise_queries '1'
	option localservice '1'
	option nohosts '1'
	option nonegcache '1'
	option noresolv '1'
	option port '1053'
	option readethers '0'
	option rebind_localhost '0'
	option rebind_protection '1'
```

Borromini, 2017-01-01:
I had added zeus to the localhost line in /etc/hosts on the router indeed (removed that now). 

```
# uci show dhcp | grep dnsmasq
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].localservice='0'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonwildcard='0'
dhcp.@dnsmasq[0].port='54'
```

Edit: I added ```nohosts```, that seems to fix it. Thanks!

closed 2017-01-01 by Borromini