openwrt/packages #3731: Transmission needs mbed TLS-support

diizzyy, 2017-01-01:
@obsy 
As mbed TLS 1.3 (PolarSSL) is EOL support for mbed TLS 2.X would be desirable. Upstream have support in tree however there's no release as of yet. Do we want to add patches or (preferably?) use a git clone for now and just spoof the reported version to the released one?

obsy, 2017-01-01:
IMO now we should use current git version

diizzyy, 2017-01-01:
Do you want me to have a look at it or can you handle it?

obsy, 2017-01-01:
I will be able to take care of it next weekend. So if you want and have time..

diizzyy, 2017-01-01:
Sounds great, thanks!

diizzyy, 2017-01-08:
@obsy Any progress?

obsy, 2017-01-08:
Wating for RC4, see https://github.com/lede-project/source/pull/657
Without this transmission with mbedtls simply does not compile.

nbd168, 2017-01-08:
RC4 is now enabled in LEDE master.

obsy, 2017-01-08:
https://github.com/openwrt/packages/pull/3795

closed 2017-01-08 by hnyman