openwrt/packages #3733: pianod is superseeded by pianod2

diizzyy, 2017-01-01:
@thess 
Can you have a look at updating to pianod2 as pianod is now deprecated and adds new features such as mbed TLS 2.X support if you think it's still something we should keep in the tree?

thess, 2017-01-01:
I would prefer to keep pianod (small footprint). I have a private version already ported to mbed TLS (I did the original PolarSSL adaption to reduce it's size). Pianod2 has too large a footprint due to additional dependencies. I will produce a stable source package release of pianod with mbed TLS which addionally has shoutcast server support and is currently in my private repo (thess/pianod-sc). Give me a couple of days to get this setup.

I will also work with Mike Brady to adapt the shairport-sync polarssl variant or will drop the polarssl variant.

diizzyy, 2017-01-01:
Great! I take it you don't like the ffmpeg dependency for pianod2?

thess, 2017-01-03:
Didn't want ffmpeg, curl, libsdl, and/or gstreamer among others. Like to keep it small and single purpose. I've been running my private version for about a year now. Closing this by #3746.

closed 2017-01-03 by thess