openwrt/packages #3735: unbound: build for target failed, sys/sysctl.h: No such file or directory

poranje, 2017-01-02:
Maintainer: @EricLuehrsen
Environment: ar71xx, archer_c7v2, LEDE master  100644 

Description:
Build under macOS of unbound with toolchain for target fails.
The only sysctl.h file under the staging_dir is staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/linux/sysctl.h and its path is not included; see below.
Guess it is an issue with configure or environment.

> OpenWrt-libtool: compile:  mips-openwrt-linux-musl-gcc -I. -I/Users/Shared/projects/lede-project/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/Users/Shared/projects/lede-project/staging_dir/target-mips_24kc_musl-1.1.15/include -I/Users/Shared/projects/lede-project/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/Users/Shared/projects/lede-project/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/Users/Shared/projects/lede-project/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include -I/Users/Shared/projects/lede-project/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/Users/Shared/projects/lede-project/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /Users/Shared/projects/lede-project/build_dir/target-mips_24kc_musl-1.1.15/unbound-1.6.0:unbound-1.6.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -c compat/getentropy_osx.c  -fPIC -DPIC -o .libs/getentropy_osx.o
compat/getentropy_osx.c:26:24: fatal error: sys/sysctl.h: No such file or directory
compilation terminated.

EricLuehrsen, 2017-01-02:
I use Ubuntu 16.04 LTS for build environment without issue for OpenWrt CC or LEDE/master. I am not familar with Mac quirks.

poranje, 2017-01-02:
Oké, still the problem occurs when building for the **target**, so it doesn’t seem related to the build of the toolchain (or host tools).
Thanks anyway. I’ll provide more feed-back if a solution is found.

@lede-dev:
maybe this problem is not specific for package/unbound but for building on macOS in general.
for details, see: https://github.com/openwrt/packages/issues/3735

Paul

> Op 2 jan. 2017, om 15:02 heeft Eric Luehrsen <notifications@github.com> het volgende geschreven:
> 
> I use Ubuntu 16.04 LTS for build environment without issue for OpenWrt CC or LEDE/master. I am not familar with Mac quirks.

EricLuehrsen, 2017-01-16:
Have you tried this again? I noticed LEDE has made changes on the way to release, where sometimes the **host** /usr/local/... stuff was being pulled in and not the installed tools version. The commit volume is intense. I wouldn't have time to read it all. I am unsure what aspects are really affected, but it is worth a try.

poranje, 2017-01-16:
Just started a new make clean && make cycle and the build failed at unbound with the same error. Current revision is r2992-5c20a4fec9.

EricLuehrsen, 2017-06-01:
@poranje, considering the last Unbound change we worked on, is this solved or just disappeared? Can we close it?

poranje, 2017-06-01:
Totally forgot about this one. I've turned to building under Linux.
That work-around suits me well,  so I've not tested building on the Mac any more.

closed 2017-06-01 by poranje