openwrt/packages #3743: gnurl needed as a separate package?

diizzyy, 2017-01-03:
@dangowrt 
Hi, is this package really needed?
Can't we just have these changes upstreamed, also a bit unsure about it's usefulness as pretty much everything else links against other ssl libs which curl supports ootb?

dangowrt, 2017-01-03:
Unfortunately the ssl lib curl uses has to be decided at compile time. 
As curl is typically compiled  without ssl support in openwrt/lede,  
having gnurl as a seperate package allows gnunet to be installed using 
binary packages without collision and yet be sure gnutls is used by 
lib{c,gn}url. Read up about the reason of having the gnurl fork on 
https://gnunet.org/gnurl

diizzyy, 2017-01-03:
I think you misunderstood my intentions. The gnutls issue doesn't really exist in OpenWRT/LEDE as far as I can tell, however I didn't notice before posting that variants of curl isn't generated by default but that shouldn't be too hard to fix. I might be missing something but are there any scenarios where OpenSSL/mbed TLS aren't "good enough" which most other packages links against in general?

dangowrt, 2017-01-03:
Even if build-variants were used instead of configuration, the resulting 
packages would be conflicting, ie libcurl-nossl and libcurl-gnutls 
cannot be installed simultanously. Another problem is that libcurl afaik 
doesn't allow to distinguish *which* ssl library is being used.
While I personally believe that mbedtls is good enough, i still want to 
follow the author's recommendation when it comes to deciding which 
crypto-lib to use for gnunet. and as gnunet already uses gnutls itself, 
it makes sense to also use gnutls for gnunet components which make use 
of libcurl or libmicrohttpd.

diizzyy, 2017-01-03:
I see, thanks for the clarification.

closed 2017-01-03 by diizzyy