openwrt/packages #3747: Removing Speex codec from tree/repo

diizzyy, 2017-01-03:
@thess @psycho-nico @tripolar 
It's 2017 and Speex is being deprecated by Xiph.
Does anyone have opinions about removing it the repo?

diizzyy, 2017-01-04:
There are very few packages that uses it and I doubt it's actually used at all.
https://github.com/openwrt/packages/search?l=Makefile&q=speex&utf8=%E2%9C%93

thess, 2017-01-04:
Fair enough - however, you should look in the **telephony** repo too.

diizzyy, 2017-01-05:
https://github.com/openwrt/telephony/pull/130
https://github.com/openwrt/telephony/pull/131

diizzyy, 2017-01-05:
https://github.com/openwrt/packages/pull/3763

tripolar, 2017-01-11:
pulseaudio also depends on libspeexdsp

diizzyy, 2017-01-11:
@tripolar 
Good catch, not sure why both soxr and libspeexdsp are needed. Documentation is well.. incomplete ;-)

thess, 2017-01-11:
I am not convinced we need to remove the speex codec entirely from our repo. I do think we should encourage package maintainers to phase it out if possible. I do see value in having it (speex) plus packages which use it retained in the repository as long as they build and are useful to users and developers.

closed 2017-02-17 by thess