openwrt/packages #3757: keepalived: broken on octeon

stintel, 2017-01-04:
Maintainer: @commodo 
Environment: LEDE r2762-ed69e93 on Octeon (Ubiquiti RSPro)

Description:
After changing /etc/config/keepalived to use my old config in /etc/keepalived/keepalived.conf, it still doesn't work as expected.

/etc/keepalived/keepalived.conf:
```
vrrp_instance VI_1 {
    state BACKUP
    interface eth0.54
    virtual_router_id 54
    use_vmac
    priority 200
    advert_int 1
    authentication {
        auth_type PASS
        auth_pass 1111
    }
    virtual_ipaddress {
        192.168.54.1/24 brd 192.168.54.255 dev eth0.54
        fe80::1/64 dev eth0.54
    }
    garp_master_delay 1
    preempt_delay 120
    notify_master "/etc/conntrackd/primary-backup.sh primary"
    notify_backup "/etc/conntrackd/primary-backup.sh backup"
    notify_fault "/etc/conntrackd/primary-backup.sh fault"
}
```

Output of `keepalived -n -f /etc/keepalived/keepalived.conf -l -D`:
```
Starting VRRP child process, pid=4597
Netlink: EOF
Netlink: EOF
Netlink: EOF
Registering Kernel netlink reflector
Registering Kernel netlink command channel
Registering gratuitous ARP shared channel
Opening file '/etc/keepalived/keepalived.conf'.
Cant find interface eth0.54 for vrrp_instance VI_1 !!!
Please define interface keyword before use_vmac keyword
Configuration error: VRRP definition must belong to an interface
Default interface eth0 does not exist and no interface specified. Skip VRRP address.
Configuration error: VRRP definition must belong to an interface
Default interface eth0 does not exist and no interface specified. Skip VRRP address.
Configuration error: VRRP definition must belong to an interface
Default interface eth0 does not exist and no interface specified. Skip VRRP address.
VRRP_Instance(VI_1) Unknown interface !
^CStopping Keepalived v1.2.19 (01/04,2017)
```

This is of course complete BS, the interface is there. Output of `ip addr show dev eth0.54`:
```
15: eth0.54@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 04:18:d6:31:38:b3 brd ff:ff:ff:ff:ff:ff
    inet 192.168.54.254/24 brd 192.168.54.255 scope global eth0.54
       valid_lft forever preferred_lft forever
    inet6 fe80::618:d6ff:fe31:38b3/64 scope link
       valid_lft forever preferred_lft forever
```

This worked fine with keepalived version 1.2.16-3. This config also works fine on my APU2 (x86).

stintel, 2017-01-04:
@commodo found the problem, you incorrectly dropped the patch 101-update-struct-msghdr.patch in e73964fa8fae94473e9046dfd8fd505206b50ab3.

commodo, 2017-01-04:
i think i dropped for version 1.3.2 and forgot to recheck when i rolled back to 1.2.19

sorry about that

PR #3758 (2017-01-04): keepalived: fix recvmsg/sendmsg on mips64 again
Maintainer: @commodo 

Commit e73964fa8fae94473e9046dfd8fd505206b50ab3 incorrectly dropped the
patch 101-update-struct-msghdr.patch. Add it again, and while add it
also add the follow-up patch that was added upstream.

Fixes #3757.
commit 28275a74 (2017-01-04): keepalived: fix recvmsg/sendmsg on mips64 again

Commit e73964fa8fae94473e9046dfd8fd505206b50ab3 incorrectly dropped the
patch 101-update-struct-msghdr.patch. Add it again, and while add it
also add the follow-up patch that was added upstream.

Fixes #3757.
closed 2017-01-04 by stintel