openwrt/packages #3767: a lot of packages with python is not built because of repeated failure on the CRC, "hash code md5, sha256, etc."

build000, 2017-01-05:
eq. 
```
ValueError: unsupported hash type md5
ERROR:root:code for hash md5 was not found
ImportError: cannot import name md5
```
...and similar messages - the problem is the serial and concerns of many packages, so perhaps you should look at the whole subsection python and/or generally the last changes in the script building.

hnyman, 2017-01-05:
> the problem is the serial and concerns of many packages, so perhaps you should look at the whole subsection python and/or generally the last changes in the script building.

Please give concrete examples about your failures. Which packages and on which platform?

LEDE buildbot sees no failures on most platforms that I checked. But in a few platforms there may be individual problems. I noticed three failures in http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9_vfpv3/packages :
    micropython, python-cryptography and python-egenix-mx-base

Have you noticed other failures? on other platforms?

And I have nothing to do with Python packages themselves, so it makes no sense to target me with @ here. Better to include python maintainer @commodo in the discussion. It is quite possible that yesterday's python changes had negative effects that the buildbot has not yet noticed.

commodo, 2017-01-05:
Will check using the link
http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a9_vfpv3/packages

build000, 2017-01-05:
micropython, python-cryptography and python-egenix-mx-base - confirm ar71xx and apm821xx.
And oher subpackages to django.

hnyman, 2017-01-05:
> micropython, python-cryptography and python-egenix-mx-base - confirm ar71xx and apm821xx. And oher subpackages to django.

I see no failures for those packages in http://downloads.lede-project.org/snapshots/faillogs/powerpc_464fp/packages/ that is the package architecture for apm821xx.

mips_24kc (for ar71xx) has not been built lately today, but is currently building.

build000, 2017-01-05:
eq, net-snmp build when subsection python is disabled - to apm821xx..."squaring the circle"...

commodo, 2017-01-06:
@build000 @hnyman ; PR added ; 
any idea if `python-gmpy` was failing before ?

i tried building all packages on the omap target and that was the only one with a failure [different than the one in this issue]
@jefferyto any idea if that one is used ?  `python-gmpy2` does not seem to fail ; if possible i'd get rid of the first one

hnyman, 2017-01-06:
> any idea if python-gmpy was failing before ?

To my knowledge, it was not failing earlier.

I regularly monitor ar71xx/mips24kc and ipq806x/arm_cortex-a15_neon-vfpv4 platforms and I do not remember seeing any python packages breaking for some time.

closed 2017-01-06 by hnyman