openwrt/packages #3772: Opus-tools: compile with FLAC support

gorgobacka, 2017-01-06:
I'm wondering why FLAC support is disabled while compiling opus-tools here:
https://github.com/openwrt/packages/blob/master/utils/opus-tools/Makefile#L40

I would like to use this feature in opus-tools on my QNAP and opened already a issues on the corresponding place (https://github.com/Entware-ng/Entware-ng/issues/500), but the devs send me to you and will only activate it when it's activated here first.

Therefor, I would like to know if there is a specific reason behind the decision to disable it? And if not, can it be activated, please?

diizzyy, 2017-01-06:
I'm going to say no unless you're will to do a build-variant which still is pretty unnecessary, you can use ffmpeg which will do this fine.

gorgobacka, 2017-01-08:
Thanks for answering. I agree to include unnecessaries is not needed. 

However, sorry, I'm not totally sure how Entware-ng and openwrt are connected.

In Entware-ng, I tried to use ffmpeg to encode opus files (using this guide https://superuser.com/questions/516806/how-to-encode-audio-with-opus-codec), but is seems ffmpeg is not able to encode opus. (Unknown encoder 'libopus')

Can you confirm that it's working with your package? (I don't know where to look at myself) If it's working with your package, I will continue to ask on the Entware-ng issues, because then it seems that my 'feature request' is on there end.

diizzyy, 2017-01-08:
@gorgobacka 
libffmpeg-custom or libffmpeg-full is what you need, ffmpeg should default to using libffmpeg-full by default even on entware from what I can tell.

gorgobacka, 2017-01-08:
@diizzyy 
Ok, thanks. When I type 'ffmpeg -version' it does not show '--enable-libopus' as it's on my Linux Mint. But I think then I will ask on the Entware-ng again and see if their are using your ffmpeg with the same compiling parameters.

gorgobacka, 2017-01-08:
Can be closed. I was using the wrong ffmpeg. Somehow an old version was in my /usr/bin/. However, thanks for your help @diizzyy ! The error was on my end.

closed 2017-01-08 by gorgobacka