openwrt/packages #3778: youtube-dl: compiling fails

D-Albers, 2017-01-06:
Maintainer: @ianchi
Environment: Mips, Archer C5/C7, LEDE 8f0dc92afde82ab0ce7b4ca7337837422e9e6201

The build process failed with the following error message. The .config is attached as config.txt.
Feel free to request missing informations. 

```
make[3]: Entering directory '/opt/lede/trunk/feeds/packages/multimedia/youtube-dl'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl:youtube-dl -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/include -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl:youtube-dl -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/usr/include -I/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/include -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/include -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include/fortify -I/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/include " LDFLAGS="-L/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/usr/lib -L/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/lib -L/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/usr/lib -L/opt/lede/trunk/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.15/lib -znow -zrelro " make -j1 -C /opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl/. AR="mips-openwrt-linux-musl-gcc-ar" AS="ccache_cc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl:youtube-dl -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="ccache_cc" GCC="ccache_cc" CXX="ccache_cxx" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl'
/usr/bin/env python devscripts/make_supportedsites.py docs/supportedsites.md
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
Traceback (most recent call last):
  File "devscripts/make_supportedsites.py", line 13, in <module>
    import youtube_dl
  File "devscripts/../youtube_dl/__init__.py", line 15, in <module>
    from .options import (
  File "devscripts/../youtube_dl/options.py", line 8, in <module>
    from .downloader.external import list_external_downloaders
  File "devscripts/../youtube_dl/downloader/__init__.py", line 3, in <module>
    from .common import FileDownloader
  File "devscripts/../youtube_dl/downloader/common.py", line 10, in <module>
    from ..utils import (
  File "devscripts/../youtube_dl/utils.py", line 28, in <module>
    import ssl
  File "/opt/lede/trunk/staging_dir/target-mips_24kc_musl-1.1.15/host/lib/python2.7/ssl.py", line 98, in <module>
    import _ssl             # if we can't import it, let the error propagate
ImportError: No module named _ssl
Makefile:66: recipe for target 'supportedsites' failed
make[4]: *** [supportedsites] Error 1
make[4]: Leaving directory '/opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl'
Makefile:45: recipe for target '/opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl/.built' failed
make[3]: *** [/opt/lede/trunk/build_dir/target-mips_24kc_musl-1.1.15/youtube-dl/.built] Error 2
make[3]: Leaving directory '/opt/lede/trunk/feeds/packages/multimedia/youtube-dl'
package/Makefile:133: recipe for target 'package/feeds/packages/youtube-dl/compile' failed
make[2]: *** [package/feeds/packages/youtube-dl/compile] Error 2

```
[config.txt](https://github.com/openwrt/packages/files/691081/config.txt)

hnyman, 2017-01-07:
Could be something similar that @commodo fixed with #3773

Buildbot finds the error, too:
http://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/youtube-dl/

(alternatively, lots of packages broke in the last mips_24kc buildbot run due to a glib2 failure (?), but I see youtube-dl as failed also elsewhere, so maybe not. e.g. http://downloads.lede-project.org/snapshots/faillogs/mipsel_74kc/packages/ )

commodo, 2017-01-07:
Damn.
Looks like the SSL lib cannot be avoided completely.

It might be needed that we must first fix in LEDE trunk first [ in the tools/libressl folder ], to re-enable the shared lib[s] from libressl.
The ssl python [built-in] extension requires libressl/openssl.

As for the hashlib errors ;  not sure either ; I do know that Python uses openssl for those algorithms.
Though, it has some built-in alternatives.

I might be able to get around to fix this on Monday ; will see if I can do anything this weekend.

D-Albers, 2017-01-07:
Just for information:
libopenssl has been activated as module as the error appear.

commodo, 2017-01-09:
did some work today on this ; 
still more to do

i found myself trying out a few changes in the LEDE build system for the host-side builds ; 
will come back with more [as i find out stuff ]

commodo, 2017-01-12:
still not fixed yet ; 
my fault for this ; i am waiting on this to appear in the main tree : https://github.com/lede-project/source/pull/694

worst case, i could add a special `Host/Compile` build rule for Python ; but I'd prefer to wait [if it's ok] 
i don't like code duplication ; 

i was hoping it would appear in the main tree sooner

D-Albers, 2017-01-12:
@commodo 

I would rate this as "work in progress". :)
Sometimes it needs its time to get all the parts together.

commodo, 2017-01-19:
Win

@D-Albers @hnyman 
youtube-dl is fixed

D-Albers, 2017-01-20:
@commodo 

Thank you for the work and invested time. :)

closed 2017-01-22 by hnyman