openwrt/packages #3779: zerotier - error build/compile

build000, 2017-01-07:
@mwarning 
make -j8 BUILD_LOG=1
cat logs/****/compile.txt

```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   128    0   128    0     0    153      0 --:--:-- --:--:-- --:--:--   153

  0     0    0     0    0     0      0      0 --:--:--  0:00:01 --:--:--     0
100  193k    0  193k    0     0  76031      0 --:--:--  0:00:02 --:--:--  227k
100 10.7M    0 10.7M    0     0  3153k      0 --:--:--  0:00:03 --:--:-- 6324k

Applying ./patches/0001-Fix-for-running-under-MUSL-libc-e.g.-Alpine-Linux.patch using plaintext: 
patching file osdep/Thread.hpp

Applying ./patches/0002-fix-build.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0003-workaround_to_avoid_seg_faults.patch using plaintext: 
patching file service/OneService.cpp

Applying ./patches/0005-use-internal-miniupnpc-and-natpmp using plaintext: 
patching file make-linux.mk
patching file osdep/PortMapper.cpp
make[4]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14'
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/C25519.o node/C25519.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/CertificateOfMembership.o node/CertificateOfMembership.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/Cluster.o node/Cluster.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/DeferredPackets.o node/DeferredPackets.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/Identity.o node/Identity.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/Incom [... cut ...]
<builtin>: recipe for target 'ext/miniupnpc/minisoap.o' failed
make[4]: *** [ext/miniupnpc/minisoap.o] Error 1
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14'
Makefile:84: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14/.built] Error 2
```

hnyman, 2017-01-07:
I compiled it ok for ipq806x before merging the pull request, so the error is likely only for some platforms.
(but I did not test all optional options)

build000, 2017-01-07:
gcc/musl problem or not matches options from compile process ?

mwarning, 2017-01-07:
I will look into the problem.

mwarning, 2017-01-07:
@hnyman: I messed up with a last minute change and a cache issue that made me believe all was well.

Package builds&runs now.

build000, 2017-01-07:
confirm: this commit https://github.com/openwrt/packages/commit/55da121c752aed133fea782775e21e11283bdd0d fixed build/compile problem (enable miniupnpc option)

LEDE r2808-b18f75c_powerpc_464fp
cat .config:
```
(...)
CONFIG_PACKAGE_zerotier=m

#
# Configuration
#
CONFIG_ZEROTIER_ENABLE_DEBUG=y
CONFIG_ZEROTIER_USE_MINIUPNPC=y
CONFIG_ZEROTIER_ENABLE_NETWORK_CONTROLLER=y
CONFIG_ZEROTIER_ENABLE_SELFTEST=y

(...)
```
make -j8 BUILD_LOG=1
cat logs/***/compile.txt:
```

Applying ./patches/0001-Fix-for-running-under-MUSL-libc-e.g.-Alpine-Linux.patch using plaintext: 
patching file osdep/Thread.hpp

Applying ./patches/0002-fix-build.patch using plaintext: 
patching file make-linux.mk

Applying ./patches/0003-workaround_to_avoid_seg_faults.patch using plaintext: 
patching file service/OneService.cpp

Applying ./patches/0005-use-internal-miniupnpc-and-natpmp using plaintext: 
patching file make-linux.mk
patching file osdep/PortMapper.cpp
make[4]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14'
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/C25519.o node/C25519.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/CertificateOfMembership.o node/CertificateOfMembership.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -fno-builtin -nostdinc++ -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include/uClibc++ -DGCC_HASCLASSVISIBILITY -Wall -fPIE -fvisibility=hidden  -Wall -Wno-unused-result -Wreorder -fPIE -fno-rtti -pthread  -DNDEBUG    -c -o node/Cluster.o node/Cluster.cpp
powerpc-openwrt-linux-musl-g++ -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14:ZeroTierOne-1.1.14 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DZT_TRACE -ggdb3  -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/P [... cut ...]
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14/ipkg-powerpc_464fp/zerotier/usr/bin/zerotier-one: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14/ipkg-powerpc_464fp/zerotier/usr/bin/zerotier-selftest: executable
Packaged contents of /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/ZeroTierOne-1.1.14/ipkg-powerpc_464fp/zerotier into /home/dom/Pulpit/Projekty/WNDR4700/bin/packages/powerpc_464fp/packages/zerotier_1.1.14-4_powerpc_464fp.ipk
```

build000, 2017-01-07:
Close this issue - problem fixed in https://github.com/openwrt/packages/commit/55da121c752aed133fea782775e21e11283bdd0d

closed 2017-01-07 by build000