openwrt/packages #3790: wireguard: delay modprobe

lucize, 2017-01-08:
can wirewuard modules be delayed at boot time ?
now it has 33 priority and the boot log will be full of:

`wireguard: The xt_hashlimit module for IPv4 is required`

raising it solves spamming the boot log

regards

yousong, 2017-01-08:
@lucize , please try following the template and at least ping the package maintainer @zorun

build000, 2017-01-09:
@zorun 
Moreover, the greatest handicap is wirequard:
1) does not control any of the separate SSID
2) does not control any of the radio (2.4GHz and 5GHz) regardless
- Not always saving radio / current must be a priority; much more important from a practical point of view, are the above-mentioned functions.

zorun, 2017-01-09:
@lucize  Thanks for the report, I had never seen this error but I will have a look

lucize, 2017-01-09:
The error is displayed until iptable modules is loaded, it delays the boot process a bit, o fast systems the line is repeated around 300 times :) 

Regards

zorun, 2017-01-09:
@lucize Could you try replacing the autoload line at https://github.com/openwrt/packages/blob/master/net/wireguard/Makefile#L99 by the following?

```
AUTOLOAD:=$(call AutoProbe,wireguard)
```

I have no idea if AutoProbe will work for wireguard (I have asked on lede-dev to know what it does exactly), but since `xt_hashlimit` uses that, it seems cleaner than increasing the priority.

lucize, 2017-01-09:
as my reply to you on lede-dev it's ok tu use autoprobe (tested)
I see only one warning

regards

yousong, 2017-01-10:
Can you please try one of the following and see if they fix the dependency issue for you?

 - Set `kernel.modprobe=/usr/sbin/modprobe` in `/etc/sysctl.conf`
 - Or, `ln -s kmodloader /sbin/modprobe`

The problem is probably caused by `depends` inside `.modinfo` section of `wireguard.ko` has no explicit dependency on `hashlimit` and path of `modprobe` mismatches sysctl setting `kernel.modprobe`

lucize, 2017-01-10:
@yousong thanks for the hints, I tried both of them and the result is the same

> [    7.706040] usbserial: USB Serial support registered for generic
[    7.706438] via_rhine: v1.10-LK1.5.1 2010-10-09 Written by Donald Becker
[    7.707259] wireguard: The xt_hashlimit module for IPv4 is required
[    7.799547] xt_time: kernel timezone is -0000
[    7.799784] usbcore: registered new interface driver cdc_ether
[    7.800162] usbcore: registered new interface driver cdc_ncm
[    7.800353] usbcore: registered new interface driver cdc_subset
[    7.801766] usbcore: registered new interface driver huawei_cdc_ncm
[    7.805302] PPP generic driver version 2.4.2
[    7.805533] PPP MPPE Compression module registered
[    7.805752] NET: Registered protocol family 24
[    7.876474] wireguard: WireGuard 0.0.20161230 loaded. See www.wireguard.io for information.
[    7.876678] wireguard: Copyright (C) 2015-2016 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
[    7.878179] usbcore: registered new interface driver option

yousong, 2017-01-10:
Well, looks like there is a missing feature in the ubox modprobe: it does not parse mod alias at the moment, e.g. the kernel calls `modprobe -- ipt_hashlimit`, but ubox modprobe is not aware of the fact that `xt_hashlimit.ko` has aliases `ipt_hashlimit` and `ip6t_hashlimit`.

Hmm, I will see if I can fix the issue this week, unless someone wants to be quicker ;)

closed 2017-01-11 by lucize

zx2c4, 2017-01-11:
@yousong you've gotten at the crux of the issue, which should be addressed. The alises for xt_hashlimit to ipt_hashlimit and ip6t_hashlimit are important for the kernel's built-in on-demand module loading to work.