openwrt/packages #3793: bcp38: implement ipv6 BCP38

bastien-roucaries, 2017-01-08:
Hi,

Could you implement BCP38 for ipv6 please ?

An example of implementation is here:
https://github.com/zec/bcp38v6

Thank you

tohojo, 2017-01-08:
Why? Ipv6 prefixes are assigned with source specific routes, so bcp38 firewall rules are redundant for ipv6...

bastien-roucaries, 2017-01-08:
according ro rfc6204 BCP38 should be implemented what why I asked.

Moreover I do not control myprovider router

Bastien

bastien-roucaries, 2017-01-08:
I means martian coming from wan will be rejected instead of dropped => saturating my ADSL line => DOS

tohojo, 2017-01-08:
Yes, but openwrt already does this because it only installs source specific default routes. So any packet with a source outside your assigned prefix will be dropped because there is no route for it. This makes the bcp38 firewall rules implemented in this package redundant for ipv6...

bastien-roucaries, 2017-01-08:
Even for multicast ?

bastien-roucaries, 2017-01-08:
At least if not needed could be documented in package description ?

tohojo, 2017-01-08:
Sure, can add a note somewhere. Should work for multicast as well :)

commit 3b73eb19 (2017-01-08): bcp38: Add note about IPv6 to package description and help text

Fixes #3793.
closed 2017-01-08 by tohojo