openwrt/packages #3808: ddns-scripts - mythic-beasts missing domain supressor

ducapedia, 2017-01-10:
The DDNS script for Mythic-Beats host, has a little issue.
When you type the hostname in the field, it wont supress the domain part to update the IP only for the A-DNS.
Instead, it is creating a A-name for a complete host.domain.

for example.
You should type hostname: myhouse.example.com
and in DNS control panel in mythical-beast website, you will see a A-name called "myhouse.example.com" instead simple "myhouse"

To fix it just add: "&origin=."   in the end of script at "/usr/lib/ddns/services" (you'll have to find the line)

The correct line it will be:

"mythic-beasts.com" "http://dnsapi4.mythic-beasts.com/?domain=[USERNAME]&password=[PASSWORD]&command=REPLACE%20[DOMAIN]%2060%20A%20DYNAMIC_IP&origin=."

chris5560, 2017-01-10:
@ducapedia
What OpenWrt/ddns-scripts version are you taking about.
Looking on the current services and services_ipv6 at https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/services and https://github.com/openwrt/packages/blob/master/net/ddns-scripts/files/services_ipv6 the extension is there.

dibdot, 2018-08-17:
no feedback - close this by now.

closed 2018-08-17 by dibdot