openwrt/packages #3824: coova-chilli - error build/compile

build000, 2017-01-12:
@kaloz @rajanvaja 

make -j8 BUILD_LOG=1
cat logs/***/compile.txt
```
make[4]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make  all-recursive
make[5]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
Making all in bstring
make[6]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/bstring'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/bstring'
Making all in src
make[6]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
Making all in linux
make[7]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src/linux'
make -C /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/linux-apm821xx_nand/linux-4.4.40 M=$PWD;
make[8]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/linux-apm821xx_nand/linux-4.4.40'
  Building modules, stage 2.
  MODPOST 1 modules
make[8]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/linux-apm821xx_nand/linux-4.4.40'
make[7]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src/linux'
make[7]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
/bin/bash ../libtool  --tag=CC   --mode=compile powerpc-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I..   -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include  -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -O2 -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF='"/etc/chilli.conf"' -DDEFPIDFILE='"/var/run/chilli.pid"' -DDEFSTATEDIR='"/var/run"' -DSBINDIR='"/usr/sbin"'    -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128:coova-chilli-1.3.0+20141128 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC  -MT options.lo -MD -MP -MF .deps/options.Tpo -c -o options.lo options.c
libtool: compile:  powerpc-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I.. -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/target-powerpc_464fp_musl-1.1.15/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/usr/include -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include/fortify -I/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/toolchain-powerpc_464fp_gcc-5.4.0_musl-1.1.15/include -D_GNU_SOURCE -Wall -Werror -fno-builtin -fno-strict-aliasing -O2 -fomit-frame-pointer -funroll-loops -pipe -I../bstring -DDEFCHILLICONF=\"/etc/chilli.conf\" -DDEFPIDFILE=\"/var/run/chilli.pid\" -DDEFSTATEDIR=\"/var/run\" -DSBINDIR=\"/usr/sbin\" -Os -pipe -mcpu=464fp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128:coova-chilli-1.3.0+20141128 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIC -MT options.lo -MD -MP -MF .deps/options.Tpo -c options.c  -fPIC -DPIC -o .libs/options.o
options.c: In function 'options_fromfd':
options.c:381:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_host.allocated)
                                                ^
options.c:383:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_path.allocated)
                                                ^
options.c:385:46: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_qs.allocated)
                                              ^
Makefile:897: recipe for target 'options.lo' failed
make[7]: *** [options.lo] Error 1
make[7]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
Makefile:944: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
Makefile:410: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
Makefile:336: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
Makefile:149: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/.built] Error 2
```

hnyman, 2017-01-12:
Note that buildbot has built coova-chilli for your platform this morning quite successfully, so it might be something specific to your build environment.
http://downloads.lede-project.org/snapshots/packages/powerpc_464fp/packages/coova-chilli_1.3.0+20141128-4_powerpc_464fp.ipk

You might need to re-test it after "make dirclean". 
And please also test compiling with -j1

build000, 2017-01-12:
@hnyman 
... It was like I fell and repeat compilation - thanks.

hnyman, 2017-01-12:
Based on your edited first post, you use non-standard coova-chilli config options, so the reason might well be one of them. Please try to find out, which option triggers the error.

```
CONFIG_COOVACHILLI_PROXY=y
CONFIG_COOVACHILLI_REDIR=y
CONFIG_COOVACHILLI_MINIPORTAL=y
CONFIG_COOVACHILLI_USERAGENT=y
CONFIG_COOVACHILLI_DNSLOG=y
CONFIG_COOVACHILLI_UAMDOMAINFILE=y
CONFIG_COOVACHILLI_LARGELIMITS=y
# CONFIG_COOVACHILLI_NOSSL is not set
# CONFIG_COOVACHILLI_MATRIXSSL is not set
# CONFIG_COOVACHILLI_CYASSL is not set
CONFIG_COOVACHILLI_OPENSSL=y
```

build000, 2017-01-12:
@hnyman 
Ok - I check it out and let you know in this issue

build000, 2017-01-12:
Identical make option (1st post #3824 (comment)) and new .config:
```
(...)
CONFIG_PACKAGE_kmod-ipt-coova=m
(...)
#
# Captive Portals
#
CONFIG_PACKAGE_coova-chilli=m

#
# Configuration
#
CONFIG_COOVACHILLI_PROXY=y
# CONFIG_COOVACHILLI_REDIR is not set
# CONFIG_COOVACHILLI_MINIPORTAL is not set
# CONFIG_COOVACHILLI_USERAGENT is not set
# CONFIG_COOVACHILLI_DNSLOG is not set
# CONFIG_COOVACHILLI_UAMDOMAINFILE is not set
# CONFIG_COOVACHILLI_LARGELIMITS is not set
CONFIG_COOVACHILLI_NOSSL=y
# CONFIG_COOVACHILLI_MATRIXSSL is not set
# CONFIG_COOVACHILLI_CYASSL is not set
# CONFIG_COOVACHILLI_OPENSSL is not set
(...)
```
 - new full compile (after "make dirclean"):
cat logs/***/compile.txt (**success**)
```

Applying ./patches/100-fix-sysinfo-redeclaration.patch using plaintext: 
patching file src/system.h

Applying ./patches/200-fix_compile_kmod.patch using plaintext: 
patching file src/linux/Makefile

Applying ./patches/201-fix_dereferencing_pointers.patch using plaintext: 
patching file src/linux/xt_coova.c

Applying ./patches/300-fix-compile-with-cyassl.patch using plaintext: 
patching file src/md5.h
patching file src/md5.c

Applying ./patches/400-fix-compile-with-musl.patch using plaintext: 
patching file src/system.h
patching file src/chilli_limits.h
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/automake line 3938.
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:15: installing './compile'
configure.in:18: installing './config.guess'
configure.in:8: installing './install-sh'
configure.in:8: installing './missing'
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
bstring/Makefile.am: installing './depcomp'
json/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:97: warning: source file 'avl/avl.c' is in a subdirectory,
src/Makefile.am:97: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
[... 669 lines cut ...]
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/distro'
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/distro'
make[5]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[6]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_proxy: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_response: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_radconfig: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_query: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_opt: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/lib/libbstring.so.0.0.0: shared object
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/lib/iptables/libxt_coova.so: shared object
[... cut ...]
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/kmod-ipt-coova/lib/modules/4.4.40/xt_coova.ko: relocatable
Packaged contents of /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/kmod-ipt-coova into /home/dom/Pulpit/Projekty/WNDR4700/bin/targets/apm821xx/nand/packages/kmod-ipt-coova_4.4.40+1.3.0+20141128-4_powerpc_464fp.ipk
```
...cdn. = to be continued = continues in next episode

build000, 2017-01-12:
Identical make option (1st post #3824 (comment)) and new .config:
```
(...)
CONFIG_PACKAGE_kmod-ipt-coova=m
(...)
#
# Captive Portals
#
CONFIG_PACKAGE_coova-chilli=m

#
# Configuration
#
CONFIG_COOVACHILLI_PROXY=y
CONFIG_COOVACHILLI_REDIR=y
# CONFIG_COOVACHILLI_MINIPORTAL is not set
# CONFIG_COOVACHILLI_USERAGENT is not set
# CONFIG_COOVACHILLI_DNSLOG is not set
# CONFIG_COOVACHILLI_UAMDOMAINFILE is not set
# CONFIG_COOVACHILLI_LARGELIMITS is not set
CONFIG_COOVACHILLI_NOSSL=y
# CONFIG_COOVACHILLI_MATRIXSSL is not set
# CONFIG_COOVACHILLI_CYASSL is not set
# CONFIG_COOVACHILLI_OPENSSL is not set
```
new full compile (after "make dirclean")
cat logs/***/compile.txt (**error**):
```

Applying ./patches/100-fix-sysinfo-redeclaration.patch using plaintext: 
patching file src/system.h

Applying ./patches/200-fix_compile_kmod.patch using plaintext: 
patching file src/linux/Makefile

Applying ./patches/201-fix_dereferencing_pointers.patch using plaintext: 
patching file src/linux/xt_coova.c

Applying ./patches/300-fix-compile-with-cyassl.patch using plaintext: 
patching file src/md5.h
patching file src/md5.c

Applying ./patches/400-fix-compile-with-musl.patch using plaintext: 
patching file src/system.h
patching file src/chilli_limits.h
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/automake line 3938.
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:15: installing './compile'
configure.in:18: installing './config.guess'
configure.in:8: installing './install-sh'
configure.in:8: installing './missing'
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
bstring/Makefile.am: installing './depcomp'
json/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:97: warning: source file 'avl/avl.c' is in a subdirectory,
src/Makefile.am:97: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
[... 356 lines cut ...]
options.c:381:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_host.allocated)
                                                ^
options.c:383:48: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_path.allocated)
                                                ^
options.c:385:46: error: 'regex_t {aka struct re_pattern_buffer}' has no member named 'allocated'
     if (_options.regex_pass_throughs[i].re_qs.allocated)
                                              ^
Makefile:897: recipe for target 'options.lo' failed
make[7]: *** [options.lo] Error 1
make[7]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
Makefile:944: recipe for target 'all-recursive' failed
make[6]: *** [all-recursive] Error 1
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/src'
Makefile:410: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
Makefile:336: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
Makefile:149: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/.built] Error 2
```
...cdn. = to be continued = continues in next episode

build000, 2017-01-12:
Identical make option (1st post #3824 (comment)) and new .config:
```
(...)
CONFIG_PACKAGE_kmod-ipt-coova=m
(...)
#
# Captive Portals
#
CONFIG_PACKAGE_coova-chilli=m

#
# Configuration
#
CONFIG_COOVACHILLI_PROXY=y
# CONFIG_COOVACHILLI_REDIR is not set
CONFIG_COOVACHILLI_MINIPORTAL=y
CONFIG_COOVACHILLI_USERAGENT=y
CONFIG_COOVACHILLI_DNSLOG=y
CONFIG_COOVACHILLI_UAMDOMAINFILE=y
CONFIG_COOVACHILLI_LARGELIMITS=y
# CONFIG_COOVACHILLI_NOSSL is not set
# CONFIG_COOVACHILLI_MATRIXSSL is not set
# CONFIG_COOVACHILLI_CYASSL is not set
CONFIG_COOVACHILLI_OPENSSL=y
(...)
```
- new full compile (after "make dirclean")
cat logs/***/compile.txt (success):
```

Applying ./patches/100-fix-sysinfo-redeclaration.patch using plaintext: 
patching file src/system.h

Applying ./patches/200-fix_compile_kmod.patch using plaintext: 
patching file src/linux/Makefile

Applying ./patches/201-fix_dereferencing_pointers.patch using plaintext: 
patching file src/linux/xt_coova.c

Applying ./patches/300-fix-compile-with-cyassl.patch using plaintext: 
patching file src/md5.h
patching file src/md5.c

Applying ./patches/400-fix-compile-with-musl.patch using plaintext: 
patching file src/system.h
patching file src/chilli_limits.h
libtoolize: putting auxiliary files in `.'.
libtoolize: copying file `./ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
libtoolize: copying file `m4/libtool.m4'
libtoolize: copying file `m4/ltoptions.m4'
libtoolize: copying file `m4/ltsugar.m4'
libtoolize: copying file `m4/ltversion.m4'
libtoolize: copying file `m4/lt~obsolete.m4'
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/automake line 3938.
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:15: installing './compile'
configure.in:18: installing './config.guess'
configure.in:8: installing './install-sh'
configure.in:8: installing './missing'
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
bstring/Makefile.am: installing './depcomp'
json/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
src/Makefile.am:97: warning: source file 'avl/avl.c' is in a subdirectory,
src/Makefile.am:97: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
[... 734 lines cut ...]
 /bin/mkdir -p '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-install/etc/chilli/www'
 /usr/bin/install -c -m 644 contact_form.tmpl email_visitor.tmpl login_map.tmpl register_form.tmpl contact_thanks.tmpl footer.tmpl login_register.tmpl register.tmpl contact.tmpl header.tmpl login_success.tmpl terms_form.tmpl css.tmpl js.tmpl login.tmpl terms.tmpl email_contact.tmpl login_footer.tmpl map.tmpl title.tmpl email_owner.tmpl login_form.tmpl openid_form.tmpl error.tmpl '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-install/etc/chilli/www'
 /bin/mkdir -p '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-install/etc/chilli/www'
 /usr/bin/install -c contact.chi login.chi map.chi register.chi terms.chi tos.chi payment.chi error.chi status.chi service.chi config-local.sh functions.sh payment.sh uam.sh wispr.sh '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-install/etc/chilli/www'
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/miniportal'
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/miniportal'
make[5]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[6]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[6]: Nothing to be done for 'install-exec-am'.
make[6]: Nothing to be done for 'install-data-am'.
make[6]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[5]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128'
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_proxy: executable
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/sbin/chilli_response: executable
[... cut ...]
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli/usr/lib/libchilli.so.0.0.0: shared object
Packaged contents of /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/coova-chilli into /home/dom/Pulpit/Projekty/WNDR4700/bin/packages/powerpc_464fp/packages/coova-chilli_1.3.0+20141128-4_powerpc_464fp.ipk
rstrip.sh: /home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/coova-chilli-1.3.0+20141128/ipkg-powerpc_464fp/kmod-ipt-coova/lib/modules/4.4.40/xt_coova.ko: relocatable
Packaged

hnyman, 2017-01-12:
> option in the config file "CONFIG_COOVACHILLI_REDIR = y" can cause errors

Based on your debug work, that sounds logical.

Hopefully somebody is interested enough in coova-chilli to fix it.

teslamint, 2017-05-24:
looks like duplicate of #2297

andysyam, 2017-09-13:
where do i save this file coova-chilli_1.3.0+20141128-4_powerpc_464fp.ipk

diizzyy, 2019-03-28:
Compiles using master branch on buildbots, closing

closed 2019-03-28 by diizzyy