openwrt/packages #3830: ffmpeg performance exploration....

diizzyy, 2017-01-12:
@thess 

Just having a look at some of our optimizations if they're still sane or just hampering performance.
For some odd reason ffmpeg can't find libopus if you disable usage of mips16, can't see any obvious reason why. I also tried to disable mips16 on linked libs without success.

Try building on lets say ar71xx and you'll see weird breakage. I'll post the rest of my findings here.

diizzyy, 2017-01-13:
On ARM (Kirkwood):
Not using --enable-small yields around 2-20% speed boost processing audio (encoding using external libs like libopus or resampling), I'd expect it to be more on video and on x86.

We should probably consider omitting this option on x86, ARM where isn't usually a concern and still leave it for MIPS, PPC where space is limited.

diizzyy, 2017-01-13:
--enable-small vs without (ARM, Kirkwood)

```
ffmpeg -i input.wav -acodec libopus -b:a 192k -vbr on -compression_level 10 sample.opus

size=    7594kB time=00:05:10.09 bitrate= 200.6kbits/s speed=1.81x
video:0kB audio:7552kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.555978%
real    2m 54.51s
user    2m 50.49s
sys     0m 3.00s

size=    7594kB time=00:05:10.09 bitrate= 200.6kbits/s speed=1.87x
video:0kB audio:7552kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.555978%
real    2m 49.70s
user    2m 45.54s
sys     0m 3.10s

ffmpeg -i input.wav -ar 48000 resampled.wav

size=   58143kB time=00:05:10.09 bitrate=1536.0kbits/s speed=14.2x
video:0kB audio:58142kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000131%
real    0m 22.34s
user    0m 21.30s
sys     0m 1.00s

size=   58143kB time=00:05:10.09 bitrate=1536.0kbits/s speed=17.7x
video:0kB audio:58142kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.000131%
real    0m 17.90s
user    0m 16.95s
sys     0m 0.92s
```

diizzyy, 2017-01-13:
@thess 
We talked about size vs performance in #lede-dev and while size is preferred we pretty much ended up at since ffmpeg is pretty large as it is doesn't matter that much more. I'm going to say that it matters on platforms like mips where flash is sparse. however ARM and x86 platforms have much more flash and/or storage so can we enable it on those platforms?

thess, 2017-01-13:
@diizzy - I hear you...

Do you have a transcript of the IRC conversation(s)? It would have been nice to have been invited to it. I don't hang out on IRC because I am not in an office and don't spend my day at a desk (any more). However, I do read/respond to email and a chance to participate in the conversation would have been appreciated.

I try to keep ffmpeg-audio-dec less than 1MB and still handle as much audio as possible.

diizzyy, 2017-01-15:
@thess 

Would something like this be unreasonable? (You get the idea, for some reason it doesn't seem work properly however) :/

```
### Don't enable --enable-small on the following platforms: 
### aarch64, arm, i686 and x86

ifeq ($(findstring aarch64,$(CONFIG_ARCH)),)
	enable_ffmpeg_small = n
endif
ifeq ($(findstring arm,$(CONFIG_ARCH)),)
	enable_ffmpeg_small = n
endif
ifeq ($(findstring i686,$(CONFIG_ARCH)),)
	enable_ffmpeg_small = n
endif
ifeq ($(findstring x86,$(CONFIG_ARCH)),)
	enable_ffmpeg_small = n
endif

ifndef enable_ffmpeg_small
FFMPEG_CONFIGURE += \
	--enable-small
endif
```

D-Albers, 2017-01-15:
I would prefer an option at the menuconfig. The storage space is not depend from the cpu architecture.

diizzyy, 2017-01-15:
Sure, I'm not sure however if it will give a good overall impression on the platforms above as performance is pretty poor ootb.

antonlacon, 2017-01-20:
Can look at hardcoded-tables too. This makes a tradeoff between memory usage and storage size, iirc.

diizzyy, 2017-01-20:
@D-Albers 
While this is technically true the above arches have no real issues with storage space and it'll give the end user a much better experience at a very little cost.

We should also somehow integrate sound/shine as it's way faster than LAME on platforms without a FPU however it is inferior when it comes to quality.

thess, 2017-02-17:
Updated ffmpeg revision, new configs, etc. See edc9824

closed 2017-02-17 by thess