openwrt/packages #3831: php-mod-filter: add this extension ?

jorismulliez, 2017-01-13:
HI,

Maintainer: ø
Environment: All

Description: Is it possible to add the filter (http://php.net/manual/fr/book.filter.php) extension to existing php packages (https://lede-project.org/packages/start?dataflt%5B0%5D=categories_%3Dlanguages---php&dataflt%5BName_pkg-page*%7E%5D=) ?

I'm new here so if I have to put this request elsewhere, simply ask me :)

Thanks for your help,

mhei, 2017-01-13:
You can easily add filter support when you compile from scratch; simply check the box in menuconfig. By default, this is not enabled - this is why packages by buildbots lacks this feature. I'm not how much this feature would add to the footprint of the packages, but IIRC this was the reason to not enable it by default.

jorismulliez, 2017-01-13:
Thanks for your reply @mhei !
The problem is that I use the library on Onion Omega 2 and not sure that I want to compile my own php version (see the related thread : https://community.onion.io/topic/1241/omega2-php-mods-add-filter/6 )

But I understand and realize that "filter" is not an "external" php mod. So we can't have a separated package :)

mhei, 2017-01-13:
It's just about to add `CONFIG_PHP7_FILTER=y` to the build configuration. The resulting packages php7-cgi, php7-cli, php7-fpm and php7-mod-intl become larger after enabling. But in summary, this is (only) about 30kB for my ARM based platform...

mhei, 2017-01-18:
Issue is resolved on Onion side, closing here.

closed 2017-01-18 by mhei

nejtr0n, 2017-04-13:
need this extension too

commit 41007068 (2017-04-13): php7: enable filter support by default (fixes #3831)

Filter support brings several useful features which many PHP applications
rely on. To make it easier for users who don't want to build own packages
enable this by default. This slightly increases the binary SAPI packages,
but since PHP is a huge monster anyway, I doubt this hurts anybody.