openwrt/packages #3832: new glib-2.50 build/compile error

build000, 2017-01-13:
LEDE (last)
Make -j8 BUILD_LOG=1
cat logs/***/compile.txt
```

Applying ./patches/001-automake-compat.patch using plaintext: 
patching file gtk-doc.make

Applying ./patches/002-gdate-Suppress-string-format-literal-warning.patch using plaintext: 
patching file glib/gdate.c
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/aclocal -I m4 -I . --force -I m4macros ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4macros'.
OpenWrt-libtoolize: linking file `m4macros/libtool.m4'
OpenWrt-libtoolize: linking file `m4macros/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4macros/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4macros/ltversion.m4'
OpenWrt-libtoolize: linking file `m4macros/lt~obsolete.m4'
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/autoconf --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/share/aclocal --force
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/autoheader --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/share/aclocal --force
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/automake --add-missing --force-missing
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/automake line 3938.
gio/Makefile.am:108: warning: source file 'gvdb/gvdb-reader.c' is in a subdirectory,
gio/Makefile.am:108: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
gio/Makefile.am:778: warning: source file 'gvdb/gvdb-builder.c' is in a subdirectory,
gio/Makefile.am:778: but option 'subdir-objects' is disabled
gio/Makefile.am:795: warning: source file 'gvdb/gvdb-builder.c' is in a subdirectory,
gio/Makefile.am:795: but option 'subdir-objects' is disabled
gio/tests/Makefile.am:237: warning: source file '$(top_srcdir)/gio/gdbusdaemon.c' is in a subdirectory,
gio/tests/Makefile.am:237: but option 'subdir-objects' is disabled
gio/tests/Makefile.am:235: warning: source file '$(top_builddir)/gio/gdbus-daemon-generated.c' is in a subdirectory,
gio/tests/Makefile.am:235: but option 'subdir-objects' is disabled
glib/Makefile.am:95: warning: source file 'deprecated/gallocator.c' is in a subdirectory,
glib/Makefile.am:95: but option 'subdir-objects' is disabled
glib/Makefile.am:95: warning: source file 'deprecated/gcache.c' is in a subdirectory,
glib/Makefile.am:95: but option 'subdir-objects' is disabled
glib/Makefile.am:95: warning: source file 'deprecated/gcompletion.c' is in a subdirectory,
glib/Makefile.am:95: but option 'subdir-objects' is disabled
glib/Makefile.am:95: warning: source file 'deprecated/grel.c' is in a subdirectory,
glib/Makefile.am:95: but option 'subdir-objects' is disabled
glib/Makefile.am:95: warning: source file 'deprecated/gthread-deprecated.c' is in a subdirectory,
glib/Makefile.am:95: but option 'subdir-objects' is disabled
glib/tests/Makefile.am:193: warning: source file '../../glib/gwakeup.c' is in a subdirectory,
glib/tests/Makefile.am:193: but option 'subdir-objects' is disabled
glib/tests/Makefile.am:229: warning: source file '../../glib/gwakeup.c' is in a subdirectory,
glib/tests/Makefile.am:229: but option 'subdir-objects' is disabled
autoreconf: Leaving directory `.'
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/aclocal -I m4 -I . --force -I m4macros ${ACLOCAL_FLAGS}
autoreconf: configure.ac: tracing
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4macros'.
OpenWrt-libtoolize: linking file `m4macros/libtool.m4'
[... 420 lines cut ...]
checking sys/event.h usability... no
checking sys/event.h presence... no
checking for sys/event.h... no
checking for FAMOpen in -lfam... no
configure: WARNING: *** FAM support will not be built (FAM library not found) ***
checking for getxattr in -lc... yes
checking sys/xattr.h usability... yes
checking sys/xattr.h presence... yes
checking for sys/xattr.h... yes
checking for XATTR_NOFOLLOW... no
checking for LIBELF... no
checking for elf_begin in -lelf... yes
checking for elf_getshdrstrndx in -lelf... yes
checking for elf_getshdrnum in -lelf... yes
checking libelf.h usability... no
checking libelf.h presence... no
checking for libelf.h... no
checking for LIBMOUNT... no
checking libmount/libmount.h usability... no
checking libmount/libmount.h presence... no
checking for libmount/libmount.h... no
configure: error: *** Could not find libmount
Makefile:109: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/host/glib-2.50.2/.configured' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WNDR4700/build_dir/target-powerpc_464fp_musl-1.1.15/host/glib-2.50.2/.configured] Error 1
```

build000, 2017-01-13:
require install  to host debian/ubuntu/deb system libmount1-dev

hnyman, 2017-01-13:
Buildbot picks up the same error:

http://downloads.lede-project.org/snapshots/faillogs/mips64_mips64/packages/glib2/host-compile.txt
http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a5/packages/glib2/host-compile.txt

```
checking libmount/libmount.h usability... no
checking libmount/libmount.h presence... no
checking for libmount/libmount.h... no
configure: error: *** Could not find libmount
Makefile:109: recipe for target '/data/bowl-builder/arm_cortex-a5/build/sdk/build_dir/target-arm_cortex-a5_musl-1.1.15_eabi/host/glib-2.50.2/.configured' failed
make[3]: *** [/data/bowl-builder/arm_cortex-a5/build/sdk/build_dir/target-arm_cortex-a5_musl-1.1.15_eabi/host/glib-2.50.2/.configured] Error 1
make[3]: Leaving directory '/data/bowl-builder/arm_cortex-a5/build/sdk/feeds/packages/libs/glib2'
```

build000, 2017-01-13:
Only or may certainly be required in this package BuildBot and this is not the time winabraku appropriate patches for LEDE / glib2 ???

jow-, 2017-01-13:
@tripolar - you likely need to pass `--enable-libmount=no` to `HOST_CONFIGURE_VARS` as well.

tripolar, 2017-01-13:
@jow seems this fixes it, should be fix in https://github.com/openwrt/packages/commit/e7e68c08ba55c2d09a314484e7f4add5c1db90fc

closed 2017-01-13 by tripolar

neonb88, 2018-04-23:
Apologies, new to Ubuntu.  What does "pass `--enable-libmount=no` to `HOST_CONFIGURE_VARS` mean?  That looks like an environment variable; are we setting `export HOST_CONFIGURE_VARS=--enable-libmount=no` before running the install?

Not sure whether  @jow- 's answer helps other people, but this is what worked for me:
`sudo apt install libmount-dev`
also see: https://askubuntu.com/questions/847418/ubuntu14-04-install-glib-2-50-0