openwrt/packages #3834: ddns-scripts Get local IP via 'interface' failed

crick-ru, 2017-01-13:
Maintainer: @chris5560 
Environment: arch mipsel_24kc, model tp-link wdr3600 v1, LEDE trunk

Description: 
If the system has ip utility, an error occurs.
In the system log file:
```
Sat Jan 14 01:33:10 2017 user.warn ddns-scripts[4412]: changeip: Get local IP via 'interface' failed - retry 1/0 in 60 seconds
```
In the logfile /var/log/ddns/changeip.log:
```
 013310  info : Starting main loop at 2017-01-14 01:33
 013310       : Detect local IP on 'interface'
 013310       : #> ip -o addr show dev l2tp-inet scope global >/var/run/ddns/changeip.dat 2>/var/run/ddns/changeip.err
 013310       : Data detected:
 013310       : : <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1456 qdisc fq_codel state UNKNOWN qlen 3\    link/ppp
inet 95.30.179.73 peer 95.30.176.1 -1
 013310  WARN : Get local IP via 'interface' failed - retry 1/0 in 60 seconds
 013410       : #> ip -o addr show dev l2tp-inet scope global >/var/run/ddns/changeip.dat 2>/var/run/ddns/changeip.err
 013410       : Data detected:
 013410       : : <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1456 qdisc fq_codel state UNKNOWN qlen 3\    link/ppp
inet 95.30.179.73 peer 95.30.176.1 -1
 013410  WARN : Get local IP via 'interface' failed - retry 2/0 in 60 seconds
```
In the console there is an error:
```
sh: closing paren expected
```
With ipconfig utility, this error does not occur.
It appears only on l2tp connections.
On normal lan and wan connections, this error does not occur.
ip util I tried from the busybox and a standalone.
The result is the same.

chris5560, 2017-01-14:
@crick-ru
could you please send me by direct mail or post here the full output of the following command:
#> ip -o addr show dev l2tp-inet scope global

There is a problem analysing the data. To construct the correct "filter" I need the full output of ip command.
Thanks for your support

crick-ru, 2017-01-15:
```
10: l2tp-inet: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1456 qdisc fq_codel state UNKNOWN qlen 3\    link/ppp
10: l2tp-inet    inet 95.30.176.51 peer 95.30.176.1/32 scope global l2tp-inet\       valid_lft forever preferred_lft forever
```

mionica, 2017-01-15:
https://forum.lede-project.org/t/bugs-in-ddns-scripts/1000
also happens for pppoe, I was suggesting
sed -i "/BROADCAST|MULTICAST/d; /inet6 f/d; s/sec//g; s/forever/-1/g; s/\/.*preferred_lft//g; s/^.*$ip_interface *//g; s,peer [0-9\./]* ,,g"

yousong, 2018-04-08:
Fixed

closed 2018-04-08 by yousong