openwrt/packages #3836: ffmpeg: configure error in 3.2.2 update

thess, 2017-01-14:
Reported by @artynet

```
getting configure error for libmp3lame

ERROR: libmp3lame >= 3.98.3 not found

please fix this...
```

thess, 2017-01-14:
Curious we have 3.99.5 in the repo. Configure must be finding local copy?
Will check into it today

thess, 2017-01-14:
@artynet - need more information to track this down. I cannot see how this can happen with an up-to-date build environment. Make sure lame-lib is correctly built / installed in your setup. Some hint about the problem may be found in `build_dir/target-.../ffmpeg-3.2.2/config.log`.

Supply the following if possible:
 - Target system for which you are building
 - Software versions of LEDE/OpenWrt release, packages, etc.
 - Steps to reproduce

diizzyy, 2017-01-15:
Not really sure how it's supposed to work in the first please because even if I select ffmpeg-full LAME doesn't get pulled/selected.

thess, 2017-01-15:
The extra "non-free" codecs must be explicitly selected to be included. If selected, ffmpeg will get built with them. BUILD_PATENTED must also be selected.

diizzyy, 2017-01-15:
Ahh, that's good to know. We can't do the same for fdk-aac?

thess, 2017-01-15:
Yes, that was what I am planning - got some other stuff to finish up first.

thess, 2017-01-20:
@artynet - Going once, twice ... ?

thess, 2017-01-25:
Closed - no response on issue and cannot reproduce.

closed 2017-01-25 by thess