openwrt/packages #3840: git-http: missing dep to ca-bundle, causing fails

SvenRoederer, 2017-01-15:
Maintainer: @tripolar 
Environment: LEDE reboot (snapshot 14.jan. 2017)

Description:
after installing git-http and using ```git fetch``` I saw the following error
> fatal: unable to access 'https://github.com/somerepo/openwrt-routing.git/': Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x3E00) PK - Read/write of file failed

This is caused by missing /etc/ssl/certs/ca-certificates.crt. Manually installing ```ca-bundle``` fixed this, so the dependency seems wrong.

No idea what's the situation for OpenWrt.

D-Albers, 2017-01-15:
I would say, that the package is missing at the mbedtls package and not at the git-fetch package. The mbedtls package seems not to work as expected without the ca-bundle.

tripolar, 2017-01-15:
i think so too. creating a dep on `ca-bundle` in git-http is unclean

closed 2017-01-15 by tripolar