openwrt/packages #3842: Providing a "lede-17.01" branch for the upcoming LEDE release

jow-, 2017-01-15:
Hello @openwrt/package-maintainers -

in preparation for the upcoming LEDE 17.01 release I'd like to create a new branch "lede-17.01" in this repository which will be used thorough the lifetime of the LEDE 17.01 release series.

The branch will serve multiple purposes:
- It allows the LEDE release to use a stable (as in not rapidly changing) code base for building packages
- It allows to selectively backport fixes into the existing LEDE release
- It allows package maintainers to focus on their work without having to worry about breaking things

LEDE will continually rebuild the "lede-17.01" feed branch and provide updated binaries thorough the life time of the 17.01 release series.

The new branch shall only receive important bug or security fixes. Major package updates should be avoided.

I am asking for your permission to...
 - create a "lede-17.01" branch in this repository
 - cherry-pick fixes, especially security related ones, from master into said branch with or without the maintainers prior consent

Package maintainers are very welcome, but not required, to help maintaining the stable release branch together with the LEDE team. Branching for LEDE 17.01 is planned on Monday, the 16th and I'd like to have the branch in place until then.

I'd appreciate to receive some short acknowledgements before proceeding here to ensure that there's a consensus among the maintainers of this repository.

Thank you for your great work on the package feeds which turns OpenWrt and LEDE into such versatile distributions! :)

tohojo, 2017-01-15:
ACK. Looking forward to the release :)

danrl, 2017-01-15:
Hoping to get my changes merged before the branching takes place. Thanks for extending to Monday, a Friday branch would have caused me some headaches. Looking forward to 17.01 release.

hnyman, 2017-01-15:
Ack. Sounds good.

thess, 2017-01-15:
ACK from me. Will also help with merges and cherry-pick backports.

jmccrohan, 2017-01-15:
ACK from me too.

jp-bennett, 2017-01-15:
ACK, sounds good, and ACK! Monday!  I know what I'll be working on this afternoon. =)

stintel, 2017-01-15:
ACK

mhei, 2017-01-15:
ACK, and thanks for all your efforts!

jp-bennett, 2017-01-15:
Will there also be a similar branch in the luci repo?    A "desync" between packages here and packages in the luci repo could be problematic.

danrl, 2017-01-15:
@jp-bennett 
> Will there also be a similar branch in the luci repo? A "desync" between packages here and packages in the luci repo could be problematic.

Had the same thought. I even looked for a similar issue in the luci repo. I support the idea of a similar branch in the luci repo. For 15.05 there was one. I guess it will be a no-brainer once the decision was made here. No need to have similar issues open in two repos for that right now.

tripolar, 2017-01-15:
ACK

yousong, 2017-01-16:
ACK

hillz2, 2017-01-17:
What does ACK mean ?

jp-bennett, 2017-01-17:
@hillz2 Short for Acknowledge.  It means "I see what you're doing, and I'm OK with it."

hillz2, 2017-01-17:
@jp-bennett oh haha I'm sorry, English is not my native language

thess, 2017-01-17:
https://en.wikipedia.org/wiki/Acknowledgement_(data_networks)

> When the ASCII code is used to communicate between computer terminals, each terminal can send an enquiry character to request the condition of the other. The receiver of this character can respond with ACK (0000110) to indicate that it is operating normally, or NAK (0010101) to indicate an error condition.

zorun, 2017-01-28:
I would like to discuss a more accurate definition of the following statement:

> The new branch shall only receive important bug or security fixes. Major package updates should be avoided.

Should we prevent any version bump?  Or allow only minor version bumps? For instance 1.7.X → 1.7.(X+1).
Should we restrict ourselves to API-compatible and/or ABI-compatible bumps?
Or even more restrictive, should we only backport bugfix patches from newer versions instead of bumping? (that's probably a bad idea, it would require a lot of work)

For instance, this version bump #3916 for libvpx looks okay to me (after reading https://github.com/webmproject/libvpx/releases/tag/v1.6.1), but it could have benefited from an explanation about why the bump is needed and why it shouldn't break anything.

mhei, 2017-01-28:
I would say, it depends :smiley: Staying API/ABI compatible should be  a major goal to not break other/depended packages. Another point to consider is how package "upstream" works: is there a "stable line"? If for example upstream is maintaining older releases and backporting security fixes etc. I think it's easy to follow this series of minor version bumps.

For your example #3916, I would say -from just reading the release tag notes you referenced- that the version bump would have been not covered by "important bugs or security fixes". But usually the package maintainer is an user of the package and should have closer understanding if the version bump is justified.

karlp, 2017-05-05:
Does this issue still need to be open?

closed 2017-05-05 by yousong