openwrt/packages #3866: net/pen: Retire due to maintainer being absent

diizzyy, 2017-01-17:
Pen doesn't compile anymore due to OpenSSL not having SSLv3 support anymore. The maintainer hasn't been active for quite some time and there are no pull requests etc so I'm going to assume this package isn't widely used or used at all. Can we move it to packages-abandoned or just remove it?

@thess @hnyman Feel free to tag more ppl who are involved.

thess, 2017-01-18:
How about update it to a 2016 version?

diizzyy, 2017-01-18:
I'm fine with that, not having a maintainer isn't the end of the world but packages lacking seems to lag a lot. behind upstream.

hnyman, 2017-01-23:
Jow's PR updated the version, so buildbot compilation should work again.

In general, the disappearing maintainers are an increasing problem.

diizzyy, 2017-01-23:
Ok, closing

closed 2017-01-23 by diizzyy

cshoredaniel, 2017-01-23:
I wonder if the maintainer model really works with this packages feed; most of the maintainer lines are legacy and not anything like recent, and most packages don't get touched for great lengths of time.  It seems to me it's be more useful to have more like Ubuntu MOTU rather than individual maintainers of specific packages. From what I've heard even Debian is considering some kind of change to the maintainer model.

thess, 2017-01-23:
@cshoredaniel - Not a bad idea.