openwrt/packages #3876: ddns-luci-app generates a 100% CPU utilization

crick-ru, 2017-01-18:
Maintainer: @chris5560 
Environment: arch mipsel_mips32, model asus wl-500gp v1, LEDE trunk

Description:
ddns-luci-app generates a 100% CPU utilization.
In htop I see the endlessly repeating the command:
```
sh -c grep -i "all_proxy" /usr/lib/libcurl.so*> / dev / null 2> & 1
```
100% CPU load remains even if I am in any other tab of luci interface.

chris5560, 2017-01-21:
@crick-ru

Never heard before about this problem.

The call to listed grep command is done when:
- once on overview page to check for hints and it's not part of the automatic update
- once on hints page to check which text to show (no automatic update enabled on this page)
- once on detail page when opening and every time you press the save or apply button

I cannot find/see your problem on my WNDR3800 and not on my x86 VirtualBox machines.
WNDR is running 15.05 with updated scripts and luci-app-ddns 
and x86 have 15.05.1 / DD trunk / LEDE trunk and LEDE 17.01

crick-ru, 2017-01-21:
@chris5560
I'm sorry, but my asus wl-500gp router died :(

tavinus, 2017-05-07:
Hi,
I am facing the same problem. I was not even sure if this was coming from ddns-luci though. It seemd like it has been like that since yesterday as well (the thing is even hot).

I killed it a few times and it always came back. After 4 kills the CPU usage went down, but I can still see it running on too. This router is behind 2 other routers, so I really doubt there is anyone else messing with it.

I am running the trunk snapshot, since 15.05.01 does not have many packages for my ASUS RT-N16.

I also spotted more than one call for that running at the same time. Not sure it should happen, on which case, maybe a lockfile? After I killed it a few times, I was able to see it running for like 10 minutes and then no more.
Maybe it realted [with the curl ssl support](https://github.com/rakshasa/rtorrent/issues/457)? He said:

> After long testing I can confirm that curl compiled with openssl support is working fine. Problem is curl with nss.

Cheers!
Gus

fer662, 2017-05-26:
Getting the same issue here. Goes away if I remove the luci-app-ddns package. Any thoughts?

tavinus, 2017-05-26:
What version of what system are you running?
Use this template to post these commands:
<pre>
```
# opkg list-installed | grep ddns
```

```
# uname -m -r -s -v -o
```

```
# cat /etc/*release
```

```
# cat /proc/cmdline
```

```
# cat /proc/cpuinfo
```
</pre>
It would be nice to keep each command inside the ``` tags.  
  
My Asus RT-N16 is currently running LEDE trunk, but I use this router for testing so I may flash OpenWRT back to it.

fer662, 2017-05-29:
```
root@OpenWrt:~# opkg list-installed | grep ddns
ddns-scripts - 2.7.6-14
ddns-scripts_no-ip_com - 2.7.6-14
luci-app-ddns - 2.4.8-2
```

```
root@OpenWrt:~# uname -m -r -s -v -o
Linux 4.4.14 #1 Thu Aug 25 09:12:09 UTC 2016 armv5tel GNU/Linux
```

```
root@OpenWrt:~# cat /etc/*release
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='Bleeding Edge'
DISTRIB_REVISION='49926'
DISTRIB_CODENAME='designated_driver'
DISTRIB_TARGET='kirkwood/generic'
DISTRIB_DESCRIPTION='OpenWrt Designated Driver 49926'
DISTRIB_TAINTS=''
```

```
root@OpenWrt:~# cat /proc/cmdline
console=ttyS0,115200 mtdparts=nand_mtd:512k(uboot)ro,128k@512k(u_env),128k@640k(s_env),26m@2m(kernel),26m@2m(rootfs)fs,26m@28m(alt_kernel),26m@28m(alt_rootfs)fs,74m@54m(syscfg) root=/dev/mtdblock4 ro rootfstype=jffs2 serial_number=12A10604232840 uuid=C88251346E312AF69802DCA2891222C9 hw_version=RGWM-C4_0GA device_mac=20:AA:4B:66:06:3C factory_date=2012/04/14 wps_pin=11779216
```

```
root@OpenWrt:~# cat /proc/cpuinfo
processor	: 0
model name	: Feroceon 88FR131 rev 1 (v5l)
BogoMIPS	: 1196.85
Features	: swp half fastmult edsp 
CPU implementer	: 0x56
CPU architecture: 5TE
CPU variant	: 0x2
CPU part	: 0x131
CPU revision	: 1

Hardware	: Marvell Kirkwood (Flattened Device Tree)
Revision	: 0000
Serial		: 0000000000000000
```

fer662, 2017-05-29:
I found out it gets a little bit better if I just uninstall libcurl, but there's still a noticeable performance increase if I uninstall luci-app-ddns

tavinus, 2017-05-29:
Hum.. so you are running OpenWRT Designated Driver trunk.  
The original OP was using LEDE trunk.  
  
I think you /proc/cmdline is showing more stuff than I expected.  
Feel free to trim (delete) where it shows possibly undesired information like serial number (or cut it all).  
  
So, if you keep ddns-scripts and configure it from cli/uci you have no problems?  
Only when using it through luci?  
  
It has been almost a month since I posted here and I am not even sure I had this solved.  
I have linked a possible curl problem above (curl + nss).  
Maybe I have just installed the openssl support and that solved it?  
  
I will try to make some tests here but it would be nice if some dev can help us.

fer662, 2017-05-30:
I don't really care about the serial number haha.

The issue is the same in the sense that the same commands are the ones causing the CPU utilization. Regardless of how I made the configuration, if the luci part is installed, I suffer from performance issues. If i uninstall ddns-luci-app, the ddns script keeps running properly as far as I can tell, so it's not a critical problem. It's just nice to be able to see the status on the UI, that's all.

tavinus, 2017-05-30:
I have a TP-Link TL-WR1043N/ND v1 running 15.05 Chaos Chalmer and it has no issues.
I have 2 different ddns being synced there and I can spam the initial page or the Dynamic DNS page without issues.  

I have also added an afraid.freedns dynamic dns on my ASUS RT-N16 and spammed the thing. I can see the grep+curl call coming up, but it never gets to 100%. It usually stays below 10% use for that process actually. Not sure what else it could be.

These are the versions I have on LEDE:
(seems to be a tiny older than yours)
```
# opkg list-installed | grep ddns
ddns-scripts - 2.7.6-13
luci-app-ddns - 2.4.8-1
```
And these are the ones I have on OpenWRT 15.05
(as expected, these are way behind)
```
opkg list-installed | grep ddns
ddns-scripts - 2.4.3-2
luci-app-ddns - 2.2.4-1
```
Maybe try getting the `luci-app-ddns - 2.4.8-1` ?
Not sure dude...

tavinus, 2017-05-30:
Make me wonder about the package: `ddns-scripts_no-ip_com`
Do you actually need that?
I mean, I don't have that installed, but I have no-ip as an option.

fer662, 2017-05-30:
you have no-ip.pl by default, but i'm using no-ip.com. Actually I think I'm using a custom config so I could perhaps remove it, I'll check that later. The scripts that keeps running over and over definitely comes from luci-app-ddns, but i haven't dug up what keeps it running.
https://github.com/openwrt/luci/blob/master/applications/luci-app-ddns/luasrc/tools/ddns.lua#L22

jow-, 2017-05-30:
It is likely triggered by a self-refreshing LuCI page, like the status index page. That ddns.lua check code looks rather invasive, wonder if those variables should get cached somehow. @chris5560 - opinions?

fer662, 2017-05-30:
Even the login page loads slowly. Of course, it might be because the process got triggered by another page before that.

fer662, 2017-05-30:
More info
´´´
has_proxy	= (has_wgetssl or has_curlpxy or has_fetch or has_bbwget)
´´´
I saw that in the offending line wgetssl is evaluated first in the or, so relying on short circuit I installed ca-certificates which makes it find it, thus making it never check for curl. The result is that the curl check doesn't show up on the processes, but the performance is still degraded all the same, so I don't really know what's causing it.

chris5560, 2017-05-30:
Sorry but I have no knowledge how Lua/LuIC is handling its modules and defined variables.
LuCI hopefully only loads/runs modules in memory currently needed for the current screen shown in the browser.
I developed the code reading code from other luci-apps and samples and "Google" and a lot of try and error.

The "informations" defined in tools/ddns.lua line 19-37 are needed for various functionality and multiple times in all modules.
That's the reason they defined in tools/ddns.lua
I found no lua functionality to get this informations from the OS.
To prevent calls to the OS each time I need the information, I defined these variables which (hopefully) cache/buffer this informations.

For luci-app-ddns this informations are needed because detail.lua shows, allows and check settings depending on available transfer programs and their compile options.
i.e. for a while, curl was by default compiled without proxy or ssl support 
or which wget is installed and used (busybox wget, GNU wget with or without SSL, uclient-fetch with or without SSL) etc.

Casual users should have an UI that allows only configurations for the installed software and their compiled options.
Experts can use logs and directly edit config files.

jow-, 2017-05-30:
@chris5560 thats fine. I guess the offending bit is `grep -i "all_proxy" /usr/lib/libcurl.so* >/dev/null 2>&1`. This command can be *very* expensive, especially on devices with slow flash chips. Is there any possible alternative to check for `all_proxy` support?

tavinus, 2017-05-31:
I tried to search for another way of checking for curl all_proxy support but I didn't even get close.  

The problem seems to be worse on my ASUS NT-16 than on my TP-LINK TL-WR1043NDv1.
  
**Check the timings:**
ASUS RT-N16 (LEDE trunk)
```
time grep -i "all_proxy" /usr/lib/libcurl.so*
/usr/lib/libcurl.so.4:all_proxy
/usr/lib/libcurl.so.4:ALL_PROXY
/usr/lib/libcurl.so.4.4.0:all_proxy
/usr/lib/libcurl.so.4.4.0:ALL_PROXY
real    0m 1.22s
user    0m 0.51s
sys     0m 0.10s
```
TP-LINK TL-WR1043NDv1 (with Chaos Calmer)
```
# time grep -i "all_proxy" /usr/lib/libcurl.so*
/usr/lib/libcurl.so.4:all_proxy
/usr/lib/libcurl.so.4:ALL_PROXY
/usr/lib/libcurl.so.4.3.0:all_proxy
/usr/lib/libcurl.so.4.3.0:ALL_PROXY
real    0m 0.21s
user    0m 0.20s
sys     0m 0.01s
```
Not sure about how much the system version counts, but the ASUS router has a LOT more RAM and FLASH (and has a 32GB pendrive with overlayfs). Still the ASUS router takes 5 to 6 times more to complete that call.  
  
It would be nice to KNOW for sure that this is not affecting more some targets (broadcom in the case).  

Also, since the grep libcurl call keeps showing up, I am pretty sure that variable declaration does to get cached.

Cheers
Gus

philjohn, 2017-08-14:
Affecting Netgear R7800 which is a Qualcomm Atheros ARM device

dword1511, 2017-10-01:
Shouldn't the script check curl's proxy capability with `curl --help | grep noproxy`? It is such a simple task and yet it consumes so much CPU time, every time luci auto-refreshes!

chris5560, 2017-10-03:
> curl --help | grep noproxy
only shows you the help about the parameter "--noproxy" but it does not indicate, if curl package of OpenWrt/LEDE was compiled with/-out proxy support.

tavinus, 2017-12-11:
Hi all! I decided to investigate a bit more.

**Note:** All tests reported were executed many times and represent the average time for that command. I tried to use my RT-N16 with extroot for the tests, since it is the slowest thing I have for running this command. Times change a lot from one machine to another and if extroot is used, but it will get faster nonetheless.

I am also using `/usr/bin/time` in a different way so we can test piped stuff correctly.

**Also, someone should check why the call is being executed twice, since that will already cut the CPU use and time by half!**

First I decided to try and stop execution as soon as we find any `all_proxy` results. 
 - Use `grep -m1` > works but will not stop loading next file
 - Pipe all to `head -1` > just one line comes (but speed may be slower?)
 
#### Default call
```
root@LEDE:~# /usr/bin/time -p sh -c 'grep -i "all_proxy" /usr/lib/libcurl.so*'
/usr/lib/libcurl.so.4:all_proxy
/usr/lib/libcurl.so.4:ALL_PROXY
/usr/lib/libcurl.so.4.4.0:all_proxy
/usr/lib/libcurl.so.4.4.0:ALL_PROXY
real 0.63
user 0.55
sys 0.05
```
#### Now getting only one line per file with `-m1`
```
root@LEDE:~# /usr/bin/time -p sh -c 'grep -m1 -i "all_proxy" /usr/lib/libcurl.so*'
/usr/lib/libcurl.so.4:all_proxy
/usr/lib/libcurl.so.4.4.0:all_proxy
real 0.51
user 0.40
sys 0.11
```
#### Now getting only first line of first file with `| head -1`
```
root@LEDE:~# /usr/bin/time -p sh -c 'grep -m1 -i "all_proxy" /usr/lib/libcurl.so* | head -1'
/usr/lib/libcurl.so.4:all_proxy
real 0.53
user 0.45
sys 0.06
```
I think `grep` is buffering each file regardless and then `head` is discarding it (making it the same or slower). We could use `stdbuff` to fix this, but that is not available in OpenWRT by default.

Then I decided to try the `strings` command mentioned by @dword1511 and this looks a lot better.
#### Just using strings
```
root@LEDE:~# time -p sh -c 'strings /usr/lib/libcurl.so* | grep -i "all_proxy"'
all_proxy
ALL_PROXY
all_proxy
ALL_PROXY
real 0.17
user 0.15
sys 0.00
```
#### Adding `-m1`
```
root@LEDE:~# time -p sh -c 'strings /usr/lib/libcurl.so* | grep -m1 -i "all_proxy"'
all_proxy
real 0.09
user 0.08
sys 0.00
```
#### And we have a winner
```sh
strings /usr/lib/libcurl.so* | grep -m1 -i "all_proxy"
```
There is no reason to use `head` since we are already getting the whole thing as a single string and getting only one result back.

With this it went from 60+ms to 9-ms in my slow box.

This is it working as a test:
```
root@LEDE:~# strings /usr/lib/libcurl.so* | grep -m1 -i "all_proxy" >/dev/null && echo "CURL PROXY PRESENT" || echo "NO CURL PROXY"
CURL PROXY PRESENT
```
I hope someone include this as I think it would at least help mitigate the problem.
To be honest I think this should be cached somehow.
Maybe use the `/tmp` folder since that stays in `ram`?

Cheers!
Gus

unixfox, 2018-01-01:
@tavinus Your patch really fixed my issue with high CPU usage while browsing on the Luci web interface.
Many thanks.

osasco, 2018-07-04:
Hello!

I was searching for a similar issue and hit this page. Same hardware here (1043NDv1).

I've tried to apply the solution above and it worked for a while... but this high load issue appeared again after some time powered on.

I've found out that using a USB drive (ext4) as "External Root" instead "External Overlay" made the router rock solid again.
[image]

========

My router is running for 35+ days, without a glitch with 3 DDNS services updated daily.
[image]
[image]

I think that there's a bug into the overlay_fs code (or it's caching behavior), and this issue happens even when using JFFS. My parents' router (TPLink WR-740v5, ar71xx, 4MB Flash, 32MB RAM, no USB) have the same high load issue when loading the Status Page after some time powered on. This didn't happened while using the Chaos Calmer release.

I don't have knowledge to investigate this by myself, but I'm available to help with anything I can do.

dibdot, 2018-09-29:
let's close dated ddns tickets without further input.

closed 2018-09-29 by dibdot