openwrt/packages #3884: sqlite3: Error downloading sqlite-autoconf-3160000.tar.gz

pniaps, 2017-01-19:
Maintainer: @champtar
Commited by: @diizzyy

Description:
File sqlite-autoconf-3160000.tar.gz cannot be downloaded because of md5 is invalid

```
~/openwrt/scripts/download.pl "~/openwrt/dl" "sqlite-autoconf-3160000.tar.gz" "5102404047054b2cec2f43463293f94dea39425d42bf386d24596ab4fac7c7ff" "" "http://www.sqlite.org/2017/"
--2017-01-19 10:45:11--  http://www.sqlite.org/2017/sqlite-autoconf-3160000.tar.gz
Resolving www.sqlite.org (www.sqlite.org)... 67.18.92.124, 2600:3c00::f03c:91ff:fe96:b959
Connecting to www.sqlite.org (www.sqlite.org)|67.18.92.124|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2506992 (2.4M) [application/x-gzip]
Saving to: 'STDOUT'

-                                                    100%[=====================================================================================================================>]   2.39M   171KB/s    in 13s

2017-01-19 10:45:25 (188 KB/s) - written to stdout [2506992/2506992]

MD5 sum of the downloaded file does not match (file: 1282b8ae301b5a7f43e301b9bdda0a67, requested: 5102404047054b2cec2f43463293f94dea39425d42bf386d24596ab4fac7c7ff) - deleting download.

```

pniaps, 2017-01-19:
I'm using this machine to compile OpenWrt
```
pedro@Aspire-E5-575:~$ uname -a
Linux Aspire-E5-575 4.4.0-59-generic #80-Ubuntu SMP Fri Jan 6 17:47:47 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
pedro@Aspire-E5-575:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.1 LTS
Release:        16.04
Codename:       xenial
```

champtar, 2017-01-19:
hi @pniaps 
what openwrt or lede version are you using?

pniaps, 2017-01-19:
It's being compiled in openwrt until version https://github.com/openwrt/openwrt/commit/1abd6030809c2d71f2d7767d47b99aff47471b95

Surely the version is very old, but other packages compile without problems after executing `make clean`

I'm compiling now, but i will delete contents in `dl` folder and try again to be sure what i'm telling.

diizzyy, 2017-01-19:
@pniaps 
Something is broken on your end....

http://git.alpinelinux.org/cgit/aports/commit/?id=da6b594c09696721b2b38a5b55c260a2e2b44f00
http://buildroot-busybox.2317881.n4.nabble.com/git-commit-sqlite-bump-to-version-3-16-td153891.html

antonlacon, 2017-01-19:
The OpenWrt core in use is too old to understand sha256 hash in place of a md5 hash. Support was put in 2016-01-16: https://dev.openwrt.org/changeset/48253

pniaps, 2017-01-20:
Thanks to all. The issue is invalid, sorry.

I've have compiled same config with an updated OpenWrt core and it works.

What confused me was some packages where compiled without errors. I think whose wich uses md5 instead of sha256 hash.

closed 2017-01-20 by pniaps