openwrt/packages #3892: transmission fails to build on trunk

diizzyy, 2017-01-21:
Maintainer: @obsy 
Environment: ar71xx, LEDE Trunk

Description:

```
make[3]: Entering directory '/home/diizzy/fresh-trunk/source/feeds/packages/net/transmission'
rm -f /home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.transmission.mbedtls_installed
(cd /home/diizzy/fresh-trunk/source/build_dir/target-mipsel_24kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/diizzy/fresh-trunk/source/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/autom4te AUTOCONF=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/autoconf AUTOMAKE=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/automake ACLOCAL=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/aclocal AUTOHEADER=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/libtoolize LIBTOOL=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/libtool M4=/home/diizzy/fresh-trunk/source/staging_dir/host/bin/m4 AUTOPOINT=true /home/diizzy/fresh-trunk/source/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/diizzy/fresh-trunk/source/staging_dir/host/share/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/host/share/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: /home/diizzy/fresh-trunk/source/staging_dir/host/bin/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/host/share/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/aclocal -I /home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/share/aclocal -I m4 -I . --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: /home/diizzy/fresh-trunk/source/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIR, `m4'.
OpenWrt-libtoolize: linking file `m4/libtool.m4'
OpenWrt-libtoolize: linking file `m4/ltoptions.m4'
OpenWrt-libtoolize: linking file `m4/ltsugar.m4'
OpenWrt-libtoolize: linking file `m4/ltversion.m4'
OpenWrt-libtoolize: linking file `m4/lt~obsolete.m4'
autoreconf: running: /home/diizzy/fresh-trunk/source/staging_dir/host/bin/autoconf --include=/home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/host/share/aclocal --include=/home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/aclocal --include=/home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/diizzy/fresh-trunk/source/staging_dir/host/share/aclocal --force
autoreconf: configure.ac: not using Autoheader
autoreconf: running: /home/diizzy/fresh-trunk/source/staging_dir/host/bin/automake --add-missing --force-missing
third-party/libb64/Makefile.am:3: warning: source file 'src/cdecode.c' is in a subdirectory,
third-party/libb64/Makefile.am:3: but option 'subdir-objects' is disabled
automake: warning: possible forward-incompatibility.
automake: At least a source file is in a subdirectory, but the 'subdir-objects'
automake: automake option hasn't been enabled.  For now, the corresponding output
automake: object file(s) will be placed in the top-level directory.  However,
automake: this behaviour will change in future Automake versions: they will
automake: unconditionally cause object files to be placed in the same subdirectory
automake: of the corresponding sources.
automake: You are advised to start using 'subdir-objects' option throughout your
automake: project, to avoid future incompatibilities.
third-party/libb64/Makefile.am:3: warning: source file 'src/cencode.c' is in a subdirectory,
third-party/libb64/Makefile.am:3: but option 'subdir-objects' is disabled
autoreconf: Leaving directory `.'
( cd /home/diizzy/fresh-trunk/source/build_dir/target-mipsel_24kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git; ./autogen.sh --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls   --enable-daemon --enable-cli --without-gtk --enable-largefile --enable-lightweight --with-crypto=polarssl)
creating libtransmission/version.h
libtoolize: $pkgltdldir is not a directory: `/home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/libtool'
autoreconf: libtoolize failed with exit status: 1
Makefile:198: recipe for target '/home/diizzy/fresh-trunk/source/build_dir/target-mipsel_24kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/.configured_yyyyyyy' failed
make[3]: *** [/home/diizzy/fresh-trunk/source/build_dir/target-mipsel_24kc_musl-1.1.16/transmission-mbedtls/transmission-2.92+git/.configured_yyyyyyy] Error 1
make[3]: Leaving directory '/home/diizzy/fresh-trunk/source/feeds/packages/net/transmission'
package/Makefile:106: recipe for target 'package/feeds/packages/transmission/compile' failed
make[2]: *** [package/feeds/packages/transmission/compile] Error 2
make[2]: Leaving directory '/home/diizzy/fresh-trunk/source'
package/Makefile:102: recipe for target '/home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/diizzy/fresh-trunk/source/staging_dir/target-mipsel_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/diizzy/fresh-trunk/source'
/home/diizzy/fresh-trunk/source/include/toplevel.mk:197: recipe for target 'world' failed
make: *** [world] Error 2

```

Not sure what's causing this :-/

obsy, 2017-01-21:
`libtoolize: $pkgltdldir is not a directory: `/home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/libtool'`
Something in build servers?

diizzyy, 2017-01-21:
http://downloads.lede-project.org/snapshots/faillogs/mipsel_74kc/packages/transmission/mbedtls/compile.txt

cshoredaniel, 2017-01-21:
On Sat, 21 Jan 2017 12:58:50 -0800
Cezary Jackiewicz <notifications@github.com> wrote:

> `libtoolize: $pkgltdldir is not a directory:
> `/home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/libtool'`
> Something in build servers?
> 

There have been some recent build system changes in LEDE core that
aren't quite right yet.

Should forward this to lede-dev.

Regards,

Daniel

cshoredaniel, 2017-01-21:
On Sat, 21 Jan 2017 12:58:50 -0800
Cezary Jackiewicz <notifications@github.com> wrote:

> `libtoolize: $pkgltdldir is not a directory:
> `/home/diizzy/fresh-trunk/source/staging_dir/hostpkg/share/libtool'`
> Something in build servers?
> 

faad2 has the same problem.

diizzyy, 2017-01-21:
jow- Suspects that it's caused by some changes in LEDE, to be continued...

jow-, 2017-01-21:
To be specific I suspect that https://github.com/lede-project/source/commit/3d71786154fbb3fcfbd652115f3e75db3f729941 is the culprit.

I suppose that stuff from the package builds clobbers the host libtool infrastructure.

jow-, 2017-01-21:
After some more investigation I traced it down to package/libtoo/ breaking host libtool infrastructure within its own InstallDev section.

It first overwrites files and then moves parts of the host libtool resources away into the target staging dir. To me it looks like it essentially abuses the host staging_dir as temporary scratch directory.

jow-, 2017-01-21:
Should be fixed by https://github.com/lede-project/source/commit/f798776188eaf1a9c0345da66a848a5cbe3cfb71

In order to solve your problem on a pre-existing tree you need to issue a `make tools/libtool/clean` followed by a general make run.

thess, 2017-01-26:
All looks good in the package downloads for both trunk and 17.01 -- Close this?

diizzyy, 2017-01-26:
Yep, closing

closed 2017-01-26 by diizzyy