openwrt/packages #3893: gcc: Native gcc on arm_cortex-a9_vfpv3 uses softfp by default, should be hardfp

recyclebin99, 2017-01-22:
Maintainer: @noblepepper
Environment: arm_cortex-a9_vfpv3, WRT1200AC, r3003-cfb3ef3 (lede-17.01 snapshot)

Description:

Native gcc for arm_cortex-a9_vfpv3 compiles code by default using "softvfp" (soft float) instead of "vfp" (hard float).  This causes a linker error, because the target libraries are built with hard float.
This can be worked around by adding "-mfloat-abi=hard" to the gcc command line, but we should expect native builds to link on the target without needing to specify this.

Steps to reproduce:
1. Create a simple "hello.c" source file on the target:
```C
#include <stdio.h>
int main()
{
  printf("Hello world\n");
}
```
2. Attempt to compile "hello.c" with default gcc settings:
```
root@lede:~/target-gcc-test# gcc -o hello hello.c
/usr/bin/ld: error: a.out uses VFP register arguments, /tmp/ccBJaEfP.o does not
/usr/bin/ld: failed to merge target specific data of file /tmp/ccBJaEfP.o
collect2: error: ld returned 1 exit status
```
3. Successfully compile and execute "hello.c" by specifying the correct float ABI:
```
root@lede:~/target-gcc-test# gcc -mfloat-abi=hard -o hello hello.c

root@lede:~/target-gcc-test# ls -l
-rwxr-xr-x    1 root     root          4724 Jan 21 18:34 hello
-rw-r--r--    1 root     root            63 Jan  6 07:14 hello.c

root@lede:~/target-gcc-test# file hello
hello: ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-armhf.so.1, not stripped

root@lede:~/target-gcc-test# ./hello
Hello world
```

More details from the target device:
```
root@lede:/etc# uname -a
Linux lede 4.4.42 #0 SMP Tue Jan 17 13:07:47 2017 armv7l GNU/Linux

root@lede:/etc# cat openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01-SNAPSHOT'
DISTRIB_REVISION='r3003-cfb3ef3'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='mvebu/generic'
DISTRIB_ARCH='arm_cortex-a9_vfpv3'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01-SNAPSHOT r3003-cfb3ef3'
DISTRIB_TAINTS='no-all'

root@lede:/etc# head -n 5 board.json | tail -n 4
        "model": {
                "id": "armada-385-linksys-caiman",
                "name": "Linksys WRT1200AC"
        },

```

noblepepper, 2017-01-24:
The Makefile tries to pull in the options that the toolchain gcc was configured with, apparently it didn't work. 

What does gcc -v show on the target?

recyclebin99, 2017-01-24:
Below is the output with '-v' added to the compile.  Before that is the output from 'gcc -dumpspecs' (attached file).

I've been trying to figure out how to modify a local specs file, but haven't wrapped my head around it yet.  I also finally have a build environment set up to see if I can be of any help.  I've built LEDE/OpenWRT before, but never had to dig into the actual details of what's happening.  

BTW - This isn't a major issue for me, but I do like the flexibility of being able to build native.  Decided to give it a try now that LEDE is closing in on a release.  Sorry that I didn't find this earlier, even though I've been experimenting with snapshots for a few months now.

**Output from "gcc -dumpspecs" :** [lede-gcc-specs.txt](https://github.com/openwrt/packages/files/725420/lede-gcc-specs.txt)

**Output from "gcc -v hello.c" :**
```
root@lede:~/target-gcc-test# gcc -v hello.c
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/lto-wrapper
Target: arm-openwrt-linux-muslgnueabi
Configured with: /build/lede-17.01-phase2/arm_cortex-a9_vfpv3/build/sdk/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-linux-gnu --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --build=x86_64-linux-gnu --host=arm-openwrt-linux-muslgnueabi --target=arm-openwrt-linux-muslgnueabi --enable-languages=c,c++ --with-bugurl=https://dev.openwrt.org/ --with-pkgversion='OpenWrt GCC 5.4.0' --enable-shared --disable-__cxa_atexit --with-default-libstdcxx-abi=gcc4-compatible --enable-target-optspace --with-gnu-ld --disable-nls --disable-libsanitizer --disable-libvtv --disable-libcilkrts --disable-libmudflap --disable-multilib --disable-libgomp --disable-libquadmath --disable-libssp --disable-decimal-float --disable-libstdcxx-pch --with-host-libstdcxx=-lstdc++ --prefix=/usr --libexecdir=/usr/lib --with-local-prefix=/usr
Thread model: posix
gcc version 5.4.0 (OpenWrt GCC 5.4.0)
COLLECT_GCC_OPTIONS='-v' '-mtls-dialect=gnu'
 /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/cc1 -quiet -v -imultilib . -imultiarch arm-linux-gnueabi hello.c -quiet -dumpbase hello.c -mtls-dialect=gnu -auxbase hello -version -o /tmp/ccnNANMO.s
GNU C11 (OpenWrt GCC 5.4.0) version 5.4.0 (arm-openwrt-linux-muslgnueabi)
        compiled by GNU C version 5.4.0, GMP version 4.3.2, MPFR version 2.4.2, MPC version 0.8.1
GGC heuristics: --param ggc-min-expand=64 --param ggc-min-heapsize=64201
ignoring nonexistent directory "/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/include"
ignoring nonexistent directory "/usr/include/arm-linux-gnueabi"
#include "..." search starts here:
#include <...> search starts here:
 /usr/include
 /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/include
End of search list.
GNU C11 (OpenWrt GCC 5.4.0) version 5.4.0 (arm-openwrt-linux-muslgnueabi)
        compiled by GNU C version 5.4.0, GMP version 4.3.2, MPFR version 2.4.2, MPC version 0.8.1
GGC heuristics: --param ggc-min-expand=64 --param ggc-min-heapsize=64201
Compiler executable checksum: 19c01645b99d6f6657484cd20e6d12d7
COLLECT_GCC_OPTIONS='-v' '-mtls-dialect=gnu'
 as -v -meabi=5 -o /tmp/ccgffHch.o /tmp/ccnNANMO.s
GNU assembler version 2.27 (arm-openwrt-linux-muslgnueabi) using BFD version (GNU Binutils) 2.27
COMPILER_PATH=/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/:/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/:/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/:/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/:/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/
LIBRARY_PATH=/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/:/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-mtls-dialect=gnu'
 /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/collect2 -plugin /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/liblto_plugin.so -plugin-opt=/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/lto-wrapper -plugin-opt=-fresolution=/tmp/ccJKfnPg.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s --eh-frame-hdr -dynamic-linker /lib/ld-musl-arm.so.1 -X -m armelf_linux_eabi /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/crt1.o /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/crti.o /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/crtbegin.o -L/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0 -L/usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../.. /tmp/ccgffHch.o -lssp_nonshared -lgcc_s -lc -lgcc_s /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/crtend.o /usr/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/crtn.o
/usr/bin/ld: error: a.out uses VFP register arguments, /tmp/ccgffHch.o does not
/usr/bin/ld: failed to merge target specific data of file /tmp/ccgffHch.o
collect2: error: ld returned 1 exit status
```

noblepepper, 2017-01-24:
When I first looked at this I thought it was a configure option for the native gcc that was missed but after seeing the configure options from your reply I realize it is likely a missing compilation flag when gcc the native gcc is built. 

I'll need to look into this in more depth.

recyclebin99, 2017-01-24:
The other thing I noticed was the CPU type is set to arm9tdmi, which is probably a default when building gcc for ARM?

Below is the beginning of the intermediate assembler file "hello.s" for the two compile command lines:

Default build ("gcc -save-temps hello.c"):
```
	.cpu arm9tdmi
	.fpu softvfp
	.eabi_attribute 20, 1
	.eabi_attribute 21, 1
	.eabi_attribute 23, 3
	.eabi_attribute 24, 1
	.eabi_attribute 25, 1
	.eabi_attribute 26, 2
	.eabi_attribute 30, 6
	.eabi_attribute 34, 0
	.eabi_attribute 18, 4
	.arm
	.syntax divided
```

Hard-float build ("gcc -save-temps -mfloat-abi=hard hello.c"):
```
	.cpu arm9tdmi
	.eabi_attribute 28, 1
	.fpu vfp
	.eabi_attribute 20, 1
	.eabi_attribute 21, 1
	.eabi_attribute 23, 3
	.eabi_attribute 24, 1
	.eabi_attribute 25, 1
	.eabi_attribute 26, 2
	.eabi_attribute 30, 6
	.eabi_attribute 34, 0
	.eabi_attribute 18, 4
	.arm
	.syntax divided
```

recyclebin99, 2017-01-24:
Looking through some LEDE image build output, the following relevant CFLAGS options are being passed to the cross-compiler (at least in the few lines I checked):
```
-mcpu=cortex-a9 -mfpu=vfpv3-d16 -mfloat-abi=hard
```

Here is the output from "/proc/cpuinfo" on my WRT1200AC:
```
processor       : 0
model name      : ARMv7 Processor rev 1 (v7l)
BogoMIPS        : 50.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x4
CPU part        : 0xc09
CPU revision    : 1

processor       : 1
model name      : ARMv7 Processor rev 1 (v7l)
BogoMIPS        : 50.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpd32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x4
CPU part        : 0xc09
CPU revision    : 1

Hardware        : Marvell Armada 380/385 (Device Tree)
Revision        : 0000
Serial          : 0000000000000000
```

noblepepper, 2017-01-24:
If you can find where gcc is built it will be good to know if the -mfloat-abi=hard is used there. My current theory is that it is not.

I use "nohup make -j1 V=99" so I can use vim on nohup.out to search for particular things, I find the gcc build by searching for "feeds/packages/devel/gcc"

recyclebin99, 2017-01-24:
The other day I built native gcc as part of the LEDE image build, but didn't specify any verbosity options.
I did save the build output at least, and after finding where native gcc was built, it looks to me like CFLAGS is being set correctly:
```
configure: creating ./config.status
config.status: creating Makefile
rm -f /mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/.configured_*
touch /mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/.configured_yyyyyyy
export SHELL="bash"; CFLAGS="-Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap /mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0:gcc-5.4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/usr/include -I/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/include -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/usr/include -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/include/fortify -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap /mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0:gcc-5.4.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/usr/include -I/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/include -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/usr/include -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/include/fortify -I/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/include " LDFLAGS="-L/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/usr/lib -L/mnt/sdb1/lede/source/build/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/lib -L/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/usr/lib -L/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib -znow -zrelro " make -C /mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0 DESTDIR="/mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/ipkg-install"  all install
make[4]: Entering directory '/mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0'
```
I assume CFLAGS should be picked up by the make sub-processes, although the build log isn't verbose enough to show it.

But elsewhere in the log there are lines like this, which have bare-bones CFLAGS options:
```
true "AR_FLAGS=rc" "CC_FOR_BUILD=gcc" "CC_FOR_TARGET=arm-openwrt-linux-muslgnueabi-gcc" "CFLAGS=-g -O2  -g -Os" "CXXFLAGS=-g -O2 -D_GNU_SOURCE -g -Os" "CFLAGS_FOR_BUILD=" "CFLAGS_FOR_TARGET=-g -O2  -g -Os" "INSTALL=/usr/bin/install -c" "INSTALL_DATA=/usr/bin/install -c -m 644" "INSTALL_PROGRAM=/usr/bin/install -c" "INSTALL_SCRIPT=/usr/bin/install -c" "LDFLAGS=" "LIBCFLAGS=-g -O2  -g -Os" "LIBCFLAGS_FOR_TARGET=-g -O2  -g -Os" "MAKE=make" "MAKEINFO=/mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/missing makeinfo --split-size=5000000 --split-size=5000000  " "SHELL=/bin/bash" "RUNTESTFLAGS=" "exec_prefix=/usr" "infodir=/usr/info" "libdir=/usr/lib" "includedir=/usr/include" "prefix=/usr" "tooldir=/usr/arm-openwrt-linux-muslgnueabi" "gxx_include_dir=/usr/include/c++/5.4.0" "AR=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ar" "AS=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/as" "LD=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld" "RANLIB=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ranlib" "NM=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/nm" "NM_FOR_BUILD=" "NM_FOR_TARGET=/mnt/sdb1/lede/source/build/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl-1.1.15_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/5.4.0/../../../../arm-openwrt-linux-muslgnueabi/bin/nm" "DESTDIR=/mnt/sdb1/lede/source/build/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/gcc-5.4.0/ipkg-install" "WERROR=" DO=all multi-do # make
```

I will take the gcc .ipk I built (and binutils/etc if necessary) and install it on my device to see what that gives.  I will also work on making a build with V=99.

Question: Do you know how I can clean out my original build of gcc/binutils without cleaning the rest of the LEDE build?  Trying to save time, but I'm still a rookie with the build system.

noblepepper, 2017-01-25:
Hold up I have a prospective fix I would like to you to try, I will send details in shortly.

[https://wiki.openwrt.org/doc/howto/build#cleaning_up](url) gives some info on cleaning but many packages don't have a clean target (gcc is one). I just rm -rf build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/gcc-5.4.0/

noblepepper, 2017-01-25:
Try this patch [gccForArmFpu.txt](https://github.com/openwrt/packages/files/728319/gccForArmFpu.txt)

Put it in the base of your buildroot and apply it with
`cd feeds/packages/; patch -p1 < ../../gccForArmFpu.txt; cd -` 
then build as usual.

I know it will produce a package you can install but I don't have hardware it will run on.

On my board I scp these packages to the /tmp directory 
```
~# ls /tmp/*.ipk
/tmp/ar_2.27-1_mipsel_24kc.ipk
/tmp/binutils_2.27-1_mipsel_24kc.ipk
/tmp/gcc_5.4.0-1_mipsel_24kc.ipk
/tmp/libbfd_2.27-1_mipsel_24kc.ipk
/tmp/libopcodes_2.27-1_mipsel_24kc.ipk
/tmp/libstdcpp_5.4.0-1_mipsel_24kc.ipk
/tmp/make_4.2.1-2_mipsel_24kc.ipk
/tmp/objdump_2.27-1_mipsel_24kc.ipk
```
 then 
`opkg install /tmp/*.ipk`

noblepepper, 2017-01-25:
Well at least this doesn't break my gcc on mipsel.

If it doesn't work for you `gcc -v` on the target will be good info again, it should now have a `--with-float=hard` on the "Configured with" line if I got the makefile logic right

recyclebin99, 2017-01-25:
I've applied your patch and have been digging into the build process while it's compiling.

I'm starting to understand how the build process works and what the configure options do.  So the patch is simply adding a configure option that sets the default "float-abi" that the native compiler will use.  Unless "--with-float=hard" is added to the configuration command line, gcc will be built with a default of soft float.

That brings up the question of what other default options the native gcc is built with.  In particular what the "-mcpu=" and "-mfpu=" values are by default.  Since those aren't specified during configuration, I'm assuming they will be something old.  The assembly output from my "hello.c" program had a CPU definition of "arm9tdmi".  It looks like CONFIG_EXTRA_GCC_CONFIG_OPTIONS is only used in the native GCC Makefile, so presumably that's what it was added for?  If so, I would think ideally we'd want to set CONFIG_EXTRA_GCC_CONFIG_OPTIONS automatically based on the target system/profile.  Then maintainers for each target could set the native gcc configuration options to match the cross-compiler gcc options used to build the target image.

Anyway, once I've got a new .ipk for my device I'll install it and report the results.

noblepepper, 2017-01-25:
There are several pages of [gcc options](https://gcc.gnu.org/onlinedocs/gcc/Invoking-GCC.html#Invoking-GCC), I know only a very small subset of them exist and understand even fewer. mcpu and mfpu are machine specific options and sometimes (often?) deprecated.

As far as CONFIG_EXTRA_GCC_CONFIG_OPTIONS it is used only in configuring gcc (both toolchain and native) but it is not set anywhere so it is always empty, probably cruft from long ago.

This Makefile tries to produce a native gcc that mimics the toolchain gcc as much as possible, I used `gcc -v` with no arguments to see the configuration options for the toolchain gcc and compared to what you reported on the arm target.

recyclebin99, 2017-01-25:
Sorry for the delay in responding.

Your patch worked as expected: ```gcc hello.c``` produced an a.out that ran with no issues on my WRT1200AC.  I also tried a ```whetstone.c``` file I found that does floating-point operations, and it compiled/ran with no issues using the native gcc built with your patch applied.

Regarding setting other configuration options, I can see this isn't something really needed, and it could be a complicated/brittle change.  I think your patch is good enough as-is.

Thank you for the help and education.

BTW - If you decide not to submit the patch to the LEDE tree (or it doesn't make it into the first LEDE release), at least I now know that adding ```-mfloat-abi=hard``` (along with any other tuning options) to the gcc command line is all that is required to get a working native compile.

noblepepper, 2017-01-25:
This actually goes to the openwrt/packages repo which both lede & openwrt use. I'll be submitting a pull request later today.

noblepepper, 2017-01-26:
Pull request is in. 

When it is accepted you can use the new version by running `./scripts/feeds update -a` before you build
If you get a cryptic error message `rm -rf feeds/packages/devel/gcc` and updating again should fix it.

Thanks for finding this and your help fixing/testing!

noblepepper, 2017-03-16:
Fixed by [3195](https://github.com/openwrt/packages/pull/3915) this issue needs to be closed

closed 2017-03-16 by Noltari