openwrt/packages #3905: strongswan: enable mediation

hailfinger, 2017-01-23:
Maintainer: @stintel 
Environment: mips_24kc, o2 Box 6431, latest LEDE git
Build tested and runtime tested on my o2 Box 6431

Description:
strongSwan supports the IKEv2 mediation extension which enables two peers behind NAT to directly connect to each other with the help of a mediation gateway. Given that IPv6 is still not available everywhere and quite a few ISPs do not hand out public IPv4 addresses anymore, establishing IPsec connections is getting harder.

IKEv2 mediation is part of charon and can't be compiled as plugin. Please note that IKEv2 mediation support does not require medcli/medsrv which are web frontends for medation management.

Patch follows.

Enable IKEv2 mediation support in strongswan.

```
diff --git a/net/strongswan/Makefile b/net/strongswan/Makefile
index 029b8bf..997df4e 100644
--- a/net/strongswan/Makefile
+++ b/net/strongswan/Makefile
@@ -380,6 +380,7 @@ CONFIGURE_ARGS+= \
 	--with-urandom-device=/dev/urandom \
 	--with-routing-table="$(call qstrip,$(CONFIG_STRONGSWAN_ROUTING_TABLE))" \
 	--with-routing-table-prio="$(call qstrip,$(CONFIG_STRONGSWAN_ROUTING_TABLE_PRIO))" \
+	--enable-mediation \
 	$(foreach m,$(PKG_MOD_AVAILABLE), \
 	  $(if $(CONFIG_PACKAGE_strongswan-mod-$(m)),--enable-$(m),--disable-$(m)) \
 	) \
```

stintel, 2017-01-30:
@hailfinger please follow https://github.com/openwrt/packages/blob/master/CONTRIBUTING.md next time.

closed 2017-01-30 by stintel

hailfinger, 2017-01-31:
Thank you for committing the patch.
Thanks for the pointer to contributing.md. I will use the package name as prefix next time I submit a patch.

stintel, 2017-01-31:
Well, I didn't really commit your patch, as it's non-trivial to get it from this issue. The correct way is to do a PR. Anyway, it was a really small change, so the delay was just me wanting to test if it broke anything on any of my devices.

hailfinger, 2017-01-31:
My apologies. Contributing.md does not mention anything about how to provide patches, it only puts forward requirements for pull requests, so I figured putting the patch together with the description and signoff into the issue would make applying the patch easier.

If that's not the preferred form, a clarification in contributing.md about how to submit patches would be highly appreciated. Thank you.

stintel, 2017-01-31:
As the project is on Github, I guess it's assumed that people follow the Github way of contributing, thus submitting a PR.

commit f84dfc1c (2017-01-30): strongswan: enable IKEv2 Mediation Extension

Closes #3905.
commit 59313e64 (2017-02-05): strongswan: enable IKEv2 Mediation Extension

Closes #3905.