openwrt/packages #3907: php7 - error compile/build (LEDE-trunk-r3079-9993d80_powerpc_464fp)

build000, 2017-01-24:
make -j8 **(error)**
make -j1 V=s
```
(...)
make[3]: Entering directory '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/feeds/packages/lang/php7'
rm -f /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/stamp/.php7_installed
(cd /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/php-7.1.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autom4te AUTOCONF=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoconf AUTOMAKE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/automake ACLOCAL=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/aclocal AUTOHEADER=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtoolize LIBTOOL=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtool M4=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/m4 AUTOPOINT=true /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal -I m4 -I . . || true; fi; );   (cd /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/php-7.1.0; rm -f aclocal.m4; if [ -f ./configure.ac ] || [ -f ./configure.in ]; then [ -d ./autom4te.cache ] && rm -rf autom4te.cache; [ -e ./config.rpath ] || ln -s /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/scripts/config.rpath ./config.rpath; touch NEWS AUTHORS COPYING ABOUT-NLS ChangeLog; AUTOM4TE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autom4te AUTOCONF=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoconf AUTOMAKE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/automake ACLOCAL=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/aclocal AUTOHEADER=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoheader LIBTOOLIZE=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtoolize LIBTOOL=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtool M4=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/m4 AUTOPOINT=true /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoreconf -v -f -i -s -B /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal -I m4 -I . . || true; fi; );
autoreconf: Entering directory `.'
autoreconf: configure.in: not using Gettext
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal -I m4 -I . --force 
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf: configure.in: tracing
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.in and
OpenWrt-libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
OpenWrt-libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoconf --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal --force
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoheader --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal --force
autoreconf: configure.in: not using Automake
autoreconf: Leaving directory `.'
autoreconf: Entering directory `.'
autoreconf: configure.in: not using Gettext
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal -I /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal -I m4 -I . --force 
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf: configure.in: tracing
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/libtoolize --force
OpenWrt-libtoolize: putting auxiliary files in `.'.
OpenWrt-libtoolize: linking file `./ltmain.sh'
OpenWrt-libtoolize: Consider adding `AC_CONFIG_MACRO_DIR([m4])' to configure.in and
OpenWrt-libtoolize: rerunning libtoolize, to keep the correct libtool macros in-tree.
OpenWrt-libtoolize: Consider adding `-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoconf --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal --force
autoreconf: running: /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/bin/autoheader --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/host/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/hostpkg/share/aclocal --include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/usr/share/aclocal --include=m4 --include=. --prepend-include=/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/host/share/aclocal --force
autoreconf: configure.in: not using Automake
autoreconf: Leaving directory `.'
(cd /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/php-7.1.0/./; if [ -x ./configure ]; then find /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target-powerpc_464fp_musl-1.1.16/php-7.1.0/ -name config.guess | xargs -r chmod u+w; find /home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/build_dir/target [... cut ...]
package/Makefile:102: recipe for target '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/staging_dir/target-powerpc_464fp_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk'
/home/dom/Pulpit/Projekty/WNDR4700-LEDE-trunk/include/toplevel.mk:197: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

NeoRaider, 2017-01-24:
Ah, it seems I caused this error, will submit a PR to fix it in a moment.

NeoRaider, 2017-01-24:
As the fix was basically a revert of a patch I did before, I directly pushed it.

commit 919c24c5 (2017-01-24): php7: fix xml2-config path to unbreak build

Partially reverts 4a984a8d6. Fixes #3907.
commit c5142334 (2017-02-05): php7: fix xml2-config path to unbreak build

Partially reverts 4a984a8d6. Fixes #3907.
closed 2017-01-24 by NeoRaider