openwrt/packages #3912: Ruby 2.4 fails to build

joker5bb, 2017-01-25:
make[5]: Entering directory '/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0/ext/bigdecimal'
compiling bigdecimal.c
In file included from ../.././include/ruby/missing.h:23:0,
                 from ../.././include/ruby/defines.h:143,
                 from ../.././include/ruby/ruby.h:36,
                 from bigdecimal.h:12,
                 from bigdecimal.c:13:
bigdecimal.h:95:1: error: static declaration of '__finite' follows non-static declaration
 finite(double)
 ^
../.././include/ruby/missing.h:105:17: note: previous declaration of '__finite' was here
 RUBY_EXTERN int finite(double);
                 ^
In file included from bigdecimal.c:13:0:
bigdecimal.h: In function '__finite':
bigdecimal.h:95:1: error: parameter name omitted
 finite(double)
 ^
In file included from ../.././include/ruby/missing.h:23:0,
                 from ../.././include/ruby/defines.h:143,
                 from ../.././include/ruby/ruby.h:36,
                 from bigdecimal.h:12,
                 from bigdecimal.c:13:
bigdecimal.h:97:19: error: 'n' undeclared (first use in this function)
     return !isnan(n) && !isinf(n);
                   ^
bigdecimal.h:97:19: note: each undeclared identifier is reported only once for each function it appears in
Makefile:279: recipe for target 'bigdecimal.o' failed
make[5]: *** [bigdecimal.o] Error 1
make[5]: Leaving directory '/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0/ext/bigdecimal'
exts.mk:183: recipe for target 'ext/bigdecimal/all' failed
make[4]: *** [ext/bigdecimal/all] Error 2
make[4]: Leaving directory '/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0'
uncommon.mk:220: recipe for target 'build-ext' failed
make[3]: *** [build-ext] Error 2
make[3]: Leaving directory '/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0'
Makefile:695: recipe for target '/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0/.built' failed
make[2]: *** [/home/user/openwrt-pineapple-tetra/build_dir/target-mips_34kc_uClibc-0.9.33.2/ruby-2.4.0/.built] Error 2
make[2]: Leaving directory '/home/user/openwrt-pineapple-tetra/feeds/packages/lang/ruby'
package/Makefile:191: recipe for target 'package/feeds/packages/ruby/compile' failed
make[1]: *** [package/feeds/packages/ruby/compile] Error 2
make[1]: Leaving directory '/home/user/openwrt-pineapple-tetra'
/home/user/openwrt-pineapple-tetra/include/toplevel.mk:181: recipe for target 'package/ruby/compile' failed
make: *** [package/ruby/compile] Error 2

joker5bb, 2017-01-25:
have to close this, ruby packages were not selected in menuconfig

closed 2017-01-25 by joker5bb

luizluca, 2017-01-25:
It happens ;-)