openwrt/packages #3919: 'net-snmp-5.7.3/agent/mibgroup' and 'ieee802dot11.h' issues on compile

LeosSire, 2017-01-26:
Maintainer: No idea of the maintainer, looked in Makefile. No answer. 
Environment: (chaos calmer 15.05, ar71xx)

Description:
I keep getting a compile issue to do with functions which i can find declared in associated .h and .c files when i go looking.

It appears to have been solved before but I can't see where. I have also downloaded the most recent packages.

Below is the output:
```
make[6]: Entering directory '/home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3/agent/mibgroup'
/bin/bash ../../libtool  --mode=compile ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup  -I../../snmplib  -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny   -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -Ulinux -Dlinux=linux  -c -o host/hr_device.lo host/hr_device.c
OpenWrt-libtool: compile:  ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup -I../../snmplib -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -Ulinux -Dlinux=linux -c host/hr_device.c  -fPIC -DPIC -o host/.libs/hr_device.o
OpenWrt-libtool: compile:  ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup -I../../snmplib -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -Ulinux -Dlinux=linux -c host/hr_device.c -o host/hr_device.o >/dev/null 2>&1
/bin/bash ../../libtool  --mode=compile ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup  -I../../snmplib  -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny   -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -Ulinux -Dlinux=linux  -c -o host/hr_disk.lo host/hr_disk.c
OpenWrt-libtool: compile:  ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup -I../../snmplib -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -Ulinux -Dlinux=linux -c host/hr_disk.c  -fPIC -DPIC -o host/.libs/hr_disk.o
OpenWrt-libtool: compile:  ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup -I../../snmplib -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -Ulinux -Dlinux=linux -c host/hr_disk.c -o host/hr_disk.o >/dev/null 2>&1
/bin/bash ../../libtool  --mode=compile ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup  -I../../snmplib  -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include/fortify -I/home/joe/openwrt/staging_dir/toolchain-mips_34kc_gcc-5.3.0_musl-1.1.16/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include/libnl-tiny   -DNETSNMP_ENABLE_IPV6 -fno-strict-aliasing -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /home/joe/openwrt/build_dir/target-mips_34kc_musl-1.1.16/net-snmp-5.7.3:net-snmp-5.7.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic  -Ulinux -Dlinux=linux  -c -o host/hr_filesys.lo host/hr_filesys.c
OpenWrt-libtool: compile:  ccache_cc -I../../include -I. -I../../agent -I../../agent/mibgroup -I../../snmplib -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/usr/include -I/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/include -I/h [... cut ...]
make[2]: *** [package/feeds/packages/net-snmp/compile] Error 2
make[2]: Leaving directory '/home/joe/openwrt'
package/Makefile:193: recipe for target '/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/joe/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/joe/openwrt'
/home/joe/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

```

stintel, 2017-01-30:
@LeosSire please give correct info, you say this is on CC but CC doesn't use musl by default and definitely doesn't have musl-1.1.16
@pprindeville is this the same issue you were having?

pprindeville, 2017-01-30:
> is this the same issue you were having?

It is.  It was fixed in LEDE by [commit d723f25](https://git.lede-project.org/d723f25.patch) which I've asked @wigyori to port to OpenWRT.

LeosSire, 2017-01-30:
Stintel, I downloaded the .config file from the same location I have downloaded working .bin file. 

I have made a little change in code hence why I want to recompile. 

After downloading the config.diff. Changing to .config and running 'menu defconfig' I used the resulting file on "make -j1 V=s" and it gave me the above.

pprindeville, 2017-01-30:
@jow-, @LeosSire: download the above commit with wget, then apply it with "git am".

Next, look in your staging directory for .../usr/include/libnl-tiny/ and delete the "linux" tree inside of it.

Now rebuild.

stintel, 2017-01-30:
@pprindeville thanks for the info. Closing.

closed 2017-01-30 by stintel

fishy-chips, 2017-06-07:
@pprindeville Sorry, if this is a silly comment as I'm new to the community, but just wanted to mention that this change has still not been brought into openwrt-dd 17.02 as I just encountered it upon compilation. What's the process of getting this change into openwrt out of curiosity.

pprindeville, 2017-06-07:
Locate this thread ([Planning v17.01.2](http://lists.infradead.org/pipermail/lede-dev/2017-May/007727.html)) in your mailbox, and reference this commit in a response to that thread.

stintel, 2017-06-08:
@pprindeville the lede-17.01 branch was branched from master after commit d723f2573af3a575c7a1c579bfdd4700d43ee798, so it's already in there.