openwrt/packages #3921: Multiple packages: compilations broken after merges in the past 60 - 90 days

JW0914, 2017-01-28:
**Maintainer:** multiple, unknown, please see annotation below
**Environment:** Ubuntu x64 (16.04 & 16.10), DD, Armada 37x/38x/XP, k4.4.14, Linksys WRT AC Series (1200 & 1900aAC/ACS)

**Description:**

Synced buildroot a few days ago and a plethora of updates were pulled from the past 2 - 3 months  and those updates broke the buildroot.  After recreating multiple buildroots, the exact same errors are still occurring, and the list of packages no longer compiling, when they compiled with no issue prior, continues to grow.  

**Annotations:**
  * These issues are directly tied to merges that occurred since October, as any package annotated did not have this issue ~9/26 (when I built my last images)
  * Packages `iftop` and `ifstat` had not previously been built, so these _may_ not be a result of the merges; with `net-snmp` I can't be sure of either way, as it's a dependency package.

_These are just the handful I had time to determine, as I've had to go through menuconfig and make an image literally every 10 or so changes to determine what packages are at fault... tedious and time consuming to say the least._
  - There's more than 10 that used to compile before and no longer compile after the merges, however I can't recall ~5 or so, and I no longer have the time to troubleshoot since I've spent 4 days recreating multiple buldroots across 4 VMs since the myriad of issues would normally point to an environmental problem, which it's now clear this is not.

**Packages failing to compile as of 1/28:**

1. anything to do with `rsync`
  i.  _no issues prior to merged updates in the past 60 - 90 days_
2. anything to do with `samba` [incl. `cifsmount`]
  i. _no issues prior to merged updates in the past 60 - 90 days_
3. `iftop`
4. `ifstat`
5. `net-snmp`
6. `nmap`
  i.  _no issues prior to merged updates in the past 60 - 90 days_
7. `gnupg`
  i.  _no issues prior to merged updates in the past 60 - 90 days_
8. I also believe there's issues with `busybox` customizations, however I can't be 100% due to the plethora of issues the merges in the past 60 - 90 days have caused.

diizzyy, 2017-01-29:
OpenWRT "main" repo is pretty much dead, most packages should work however most (read all) packages are only tested against upstream LEDE tree.

I can assure you that rsync, samba, nmap works fine as I compiled those ~2 days ago against the 17.01 tree.

I'm not sure exactly what Linksys ARM-boxes uses off-hand but looking at the faillogs (in general) they all seems to work fine.
http://downloads.lede-project.org/snapshots/faillogs/arm_cortex-a8_vfpv3/packages/

JW0914, 2017-01-29:
The OpenWrt Repo isn't dead... as I stated above, there were well over a hundred merges that occurred in the past 60 - 90 days and there were never issues with the annotated packages in the last 2 years, so these issues are definitely due to the aforementioned merges.  

**Additionally:**
  * Unless things have changed in the past month or so, LEDE's merging back into OpenWrt.

  * Stating "_I can assure you that rsync, samba, nmap works fine as I compiled those ~2 days ago against the 17.01 tree_" doesn't make a whole lot of sense since I repeatedly stated the opposite above.    
    * You obviously did not attempt to replicate, so you're implying either it's user error on my end, or I simply have no clue what I'm talking about... _please think about what you say before you say it_.
      * Because issues on this magnitude are normally a result of environmental factors [_user error, host OS, etc_], I recreated my buildroot more than six times across 4 different VMs, each of which contained a fresh install of Ubuntu (twice for Ubuntu 16.04 and twice Ubuntu 16.10)
    * Toolchains matter, which is why I assume the environment is requested in the issue template, and makes your statement even more baffling.

**Pertaining to Diffs and Logs:**
  - _Diffs are compared against a working config that builds successfully_
  - _All download errors below can be accessed successfully in a web browser on the host machine via HTTP and FTP_

  - **_Again, to be clear, NONE of these issues existed PRIOR to the plethora of merges that occurred in the past 60 - 90 days_**

**Rsync**
  1. **_Diff:_**
`4211,4214c4211  \\ < CONFIG_PACKAGE_rsync=y \\ < # CONFIG_RSYNC_xattr is not set \\ < # CONFIG_RSYNC_acl is not set  \\ < CONFIG_RSYNC_zlib=y \\ --- \\ > # CONFIG_PACKAGE_rsync is not set
`

  2.  **_Log:_**
> make[3]: Entering directory '/home/jw0914/openwrt/dev/feeds/packages/net/rsync'
mkdir -p /home/jw0914/openwrt/dev/dl
SHELL= flock /home/jw0914/openwrt/dev/tmp/.rsync-3.1.2.tar.gz.flock -c '	/home/jw0914/openwrt/dev/scripts/download.pl "/home/jw0914/openwrt/dev/dl" "rsync-3.1.2.tar.gz" "0f758d7e000c0f7f7d3792610fad70cb" "" "http://rsync.samba.org/ftp/rsync/src"'
--2017-01-29 14:26:14--  http://rsync.samba.org/ftp/rsync/src/rsync-3.1.2.tar.gz
Resolving rsync.samba.org (rsync.samba.org)... 144.76.82.156, 2a01:4f8:192:486::443:2
Connecting to rsync.samba.org (rsync.samba.org)|144.76.82.156|:80... failed: Connection refused.
Connecting to rsync.samba.org (rsync.samba.org)|2a01:4f8:192:486::443:2|:80... failed: Network is unreachable.
Download failed.
--2017-01-29 14:26:14--  http://mirror2.openwrt.org/sources/rsync-3.1.2.tar.gz
Resolving mirror2.openwrt.org (mirror2.openwrt.org)... 46.4.11.11
Connecting to mirror2.openwrt.org (mirror2.openwrt.org)|46.4.11.11|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-01-29 14:26:15 ERROR 404: Not Found.
Download failed.
--2017-01-29 14:26:15--  http://downloads.openwrt.org/sources/rsync-3.1.2.tar.gz
Resolving downloads.openwrt.org (downloads.openwrt.org)... 78.24.191.177
Connecting to downloads.openwrt.org (downloads.openwrt.org)|78.24.191.177|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-01-29 14:26:15 ERROR 404: Not Found.
Download failed.
No more mirrors to try - giving up.
Makefile:103: recipe for target '/home/jw0914/openwrt/dev/dl/rsync-3.1.2.tar.gz' failed
make[3]: *** [/home/jw0914/openwrt/dev/dl/rsync-3.1.2.tar.gz] Error 2
make[3]: Leaving directory '/home/jw0914/openwrt/dev/feeds/packages/net/rsync'
package/Makefile:196: recipe for target 'package/feeds/packages/rsync/compile' failed
make[2]: *** [package/feeds/packages/rsync/compile] Error 2
make[2]: Leaving directory '/home/jw0914/openwrt/dev'
package/Makefile:193: recipe for target '/home/jw0914/openwrt/dev/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/jw0914/openwrt/dev/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jw0914/openwrt/dev'
/home/jw0914/openwrt/dev/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

**Samba**
  1. **_Diff:_**
`4928,4929c4928  \\ < CONFIG_PACKAGE_samba36-server=y \\ < CONFIG_PACKAGE_SAMBA_MAX_DEBUG_LEVEL=-1 \\ --- \\ > # CONFIG_PACKAGE_samba36-server is not set
`

  2. **_Log:_**
>make[3]: Entering directory '/home/jw0914/openwrt/dev/package/network/services/samba36'
mkdir -p /home/jw0914/openwrt/dev/dl
SHELL= flock /home/jw0914/openwrt/dev/tmp/.samba-3.6.25.tar.gz.flock -c '	/home/jw0914/openwrt/dev/scripts/download.pl "/home/jw0914/openwrt/dev/dl" "samba-3.6.25.tar.gz" "76da2fa64edd94a0188531e7ecb27c4e" "" "http://ftp.samba.org/pub/samba" "http://ftp.samba.org/pub/samba/stable"'
--2017-01-29 14:20:52--  http://ftp.samba.org/pub/samba/samba-3.6.25.tar.gz
Resolving ftp.samba.org (ftp.samba.org)... 144.76.82.156, 2a01:4f8:192:486::443:2
Connecting to ftp.samba.org (ftp.samba.org)|144.76.82.156|:80... failed: Connection refused.
Connecting to ftp.samba.org (ftp.samba.org)|2a01:4f8:192:486::443:2|:80... failed: Network is unreachable.
Download failed.
--2017-01-29 14:20:52--  http://ftp.samba.org/pub/samba/stable/samba-3.6.25.tar.gz
[... 36 lines cut ...]
rm -rf /home/jw0914/openwrt/dev/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/root-mvebu/tmp-gnupg
[... cut ...]
make[2]: *** [package/feeds/packages/gnupg/compile] Error 2
make[2]: Leaving directory '/home/jw0914/openwrt/dev'
package/Makefile:193: recipe for target '/home/jw0914/openwrt/dev/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/jw0914/openwrt/dev/staging_dir/target-arm_cortex-a9+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/jw0914/openwrt/dev'
/home/jw0914/openwrt/dev/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2

hnyman, 2017-01-29:
Well, at least that rsync and cifs-utils error was easily solved. Nothing to do with any "merge in 60-90 days", as that was a download error (as your log showed) and the download address had not been changed for ages. Apparently samba.org has started to enforce https-only downloads in the past months, so that http downloads get blocked without the usual automatic redirection to https.

That can be easily noticed also from console:
```
perus@ub1610:~$ wget http://rsync.samba.org/ftp/rsync/src/rsync-3.1.2.tar.gz
--2017-01-29 17:20:21--  http://rsync.samba.org/ftp/rsync/src/rsync-3.1.2.tar.gz
Resolving rsync.samba.org (rsync.samba.org)... 144.76.82.156
Connecting to rsync.samba.org (rsync.samba.org)|144.76.82.156|:80... failed: Connection refused.
perus@ub1610:~$
perus@ub1610:~$ wget https://rsync.samba.org/ftp/rsync/src/rsync-3.1.2.tar.gz
--2017-01-29 17:24:47--  https://rsync.samba.org/ftp/rsync/src/rsync-3.1.2.tar.gz
Resolving rsync.samba.org (rsync.samba.org)... 144.76.82.156
Connecting to rsync.samba.org (rsync.samba.org)|144.76.82.156|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 892724 (872K) [application/gzip]
Saving to: ‘rsync-3.1.2.tar.gz’

rsync-3.1.2.tar.gz  100%[===================>] 871,80K  2,85MB/s    in 0,3s    

2017-01-29 17:24:47 (2,85 MB/s) - ‘rsync-3.1.2.tar.gz’ saved [892724/892724]
```
I updated the download address to their current location with https://github.com/openwrt/packages/commit/eb0f2f0021e471f8b19a77a80080ca60d9ce5ed0

-------------
Of your other examples, at least iftop, ifstat, gnupg have got compiled ok for mvebu by the Openwrt buildbot a few days ago, so hard to say where your errors are coming.

Example gnupg:
https://downloads.openwrt.org/snapshots/trunk/mvebu/generic/packages/packages/
gnupg-utils_1.4.21-1_mvebu.ipk	12742	22-Jan-2017 21:09
gnupg_1.4.21-1_mvebu.ipk	327203	22-Jan-2017 21:09

I didn't check all of them.

You might be better off by opening package-specific tickets.

JW0914, 2017-01-29:
awesome, thank you

closed 2017-01-29 by JW0914