openwrt/packages #3929: perl compile (host compile) error in 17.01.0-rc1

build000, 2017-01-30:
@Naoir @hnyman 

commit = branch 17.01.0-rc1
make -j8 BUILD_LOG=1 **(error)**
cat/***/host-compile.txt:
```

Applying ./patches/010-musl-compat.patch using plaintext: 
File pp.c is read-only; trying to patch anyway
patching file pp.c

Applying ./patches/020-host-libc-dir-moved-debian+derivatives.patch using plaintext: 
File hints/linux.sh is read-only; trying to patch anyway
patching file hints/linux.sh

Applying ./patches/100-fix-cross-compile-endianness-detection.patch using plaintext: 
File config_h.SH is read-only; trying to patch anyway
patching file config_h.SH

Applying ./patches/110-always_use_miniperl.patch using plaintext: 
File Makefile.SH is read-only; trying to patch anyway
patching file Makefile.SH

Applying ./patches/300-add-relink-hack.patch using plaintext: 
patching file relink/Makefile.PL

Applying ./patches/310-recompile-hack.patch using plaintext: 
File Makefile.SH is read-only; trying to patch anyway
patching file Makefile.SH

Applying ./patches/700-tie_fetch_count_t-handle_missing_crypt.patch using plaintext: 
File t/op/tie_fetch_count.t is read-only; trying to patch anyway
patching file t/op/tie_fetch_count.t

Applying ./patches/710-threads_join-skip_ps_on_busybox.patch using plaintext: 
File dist/threads/t/join.t is read-only; trying to patch anyway
patching file dist/threads/t/join.t

Applying ./patches/900-CVE-2015-8607.patch using plaintext: 
File dist/PathTools/Cwd.xs is read-only; trying to patch anyway
patching file dist/PathTools/Cwd.xs
File dist/PathTools/t/taint.t is read-only; trying to patch anyway
patching file dist/PathTools/t/taint.t

Applying ./patches/901-CVE-2016-2381.patch using plaintext: 
File perl.c is read-only; trying to patch anyway
patching file perl.c
 
Beginning of configuration questions for perl5.
 
Checking echo to see how to suppress newlines...
...using \c
The star should be here-->*
 
First let's make sure your kit is complete.  Checking...
Looks good...
 

This installation shell script will examine your system and ask you questions
to determine how the perl5 package should be installed. If you get
stuck on a question, you may use a ! shell escape to start a subshell or
execute a command.  Many of the questions will have default answers in square
brackets; typing carriage return will give you the default.

On some of the questions which ask for file or directory names you are allowed
to use the ~name construct to specify the login directory belonging to "name",
even if you don't have a shell which knows about that.  Questions where this is
allowed will be marked "(~name ok)".

[Type carriage return to continue]  
[... 2342 lines cut ...]
Finding dependencies for opmini.o.
Finding dependencies for perlmini.o.
Updating makefile...
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/perl/perl-5.22.1'
Now you must run 'make'.

If you compile perl5 on a different machine or from a different object
directory, copy the Policy.sh file from this object directory to the
new one before you run Configure -- this will help you with most of
the policy defaults.

make[4]: Entering directory '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/perl/perl-5.22.1'
echo @`sh  cflags "optimize='-O2'" perlmini.o`  -DPERL_IS_MINIPERL -DPERL_EXTERNAL_GLOB perlmini.c
@cc -c -DPERL_CORE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=c89 -O2 -Wall -Werror=declaration-after-statement -Wextra -Wc++-compat -Wwrite-strings -DPERL_IS_MINIPERL -DPERL_EXTERNAL_GLOB perlmini.c
perl.c: In function 'S_Internals_V':
perl.c:1800:5: error: environment variable SOURCE_DATE_EPOCH must expand to a non-negative integer less than or equal to 253402300799
     PUSHs(Perl_newSVpvn_flags(aTHX_
     ^
makefile:317: recipe for target 'perlmini.o' failed
make[4]: *** [perlmini.o] Error 1
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/perl/perl-5.22.1'
Makefile:143: recipe for target '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/perl/perl-5.22.1/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/perl/perl-5.22.1/.built] Error 2
```
nedded to lm-sensors (WNDR4700)

hnyman, 2017-01-30:
> environment variable SOURCE_DATE_EPOCH must expand to a non-negative integer less than or equal to 
253402300799

That sounds like a build environment error on your build system. 

You might try setting it manually.

You might read https://forum.lede-project.org/t/lede-compile-fail-mkyaffs-problem/203 for hints.

build000, 2017-01-30:
how set manually ???
pleas example...

hnyman, 2017-01-30:
Like I wrote in your #3931 where you see the same problem, 
you might try -j 1 instead of -j 8.
That might be parallel build issue (although a bit unlikely).

> how set manually ???
> pleas example...

I already gave examples in the discussion thread:
> > You might read https://forum.lede-project.org/t/lede-compile-fail-mkyaffs-problem/203 for hints.

build000, 2017-01-30:
ok. using simple fix:
```
dom@slawek:~/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db$ date +%s > version.date
dom@slawek:~/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db$ SOURCE_DATE_EPOCH=`date +%s`
dom@slawek:~/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db$ scripts/get_source_date_epoch.sh
1485804936
dom@slawek:~/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db$ make menuconfig
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Collecting package info: done
Collecting target info: done

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

dom@slawek:~/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db$ make -j8 BUILD_LOG=1
(...)
done
```

closed 2017-01-30 by build000