openwrt/packages #3931: python comile/build (host-compile) in 17.01.0-rc1

build000, 2017-01-30:
commit = branch 17.01.0-rc1 = r3042-ec095b5
feeds.conf.default:
```
src-git packages https://git.lede-project.org/feed/packages.git^31d89be9e69bac261bfe7440512cb4e0f3356255
src-git luci https://git.lede-project.org/project/luci.git^472dc4b9e2ca71c114f5da70cb612c1089b8daa7
src-git routing https://git.lede-project.org/feed/routing.git^a6c7413594a0e4b42dab42bb5fa68534e39b7d0c
src-git telephony https://git.lede-project.org/feed/telephony.git^1f0fb2538ba6fc306198fe2a9a4b976d63adb304
```
make -j8 BUILD_LOG=1 (error)
cat logs/***/host-compile.txt:
```
Applying ./patches/001-enable-zlib.patch using plaintext: 
patching file Modules/Setup.dist
Hunk #1 succeeded at 464 (offset 106 lines).

Applying ./patches/002-do-not-add-include-dirs-when-cross-compiling.patch using plaintext: 
patching file setup.py
Hunk #1 succeeded at 497 (offset 17 lines).

Applying ./patches/003-do-not-compile-tests-at-build.patch using plaintext: 
patching file Makefile.pre.in
Hunk #1 succeeded at 1116 (offset 78 lines).
Hunk #2 succeeded at 1143 (offset 78 lines).

Applying ./patches/004-do-not-write-bytes-codes.patch using plaintext: 
patching file Python/pythonrun.c
Hunk #2 succeeded at 185 (offset 11 lines).

Applying ./patches/006-remove-debian-multiarch-support.patch using plaintext: 
patching file setup.py
Hunk #1 succeeded at 460 (offset 16 lines).

Applying ./patches/007-distutils-do-not-adjust-path.patch using plaintext: 
patching file Lib/distutils/command/build_scripts.py

Applying ./patches/008-distutils-use-python-sysroot.patch using plaintext: 
patching file Lib/distutils/sysconfig.py
patching file Lib/distutils/command/build_ext.py
Hunk #1 succeeded at 240 (offset 3 lines).

Applying ./patches/009-do-not-use-dblib_dir-when-cross-compiling.patch using plaintext: 
patching file setup.py
Hunk #1 succeeded at 1089 (offset 22 lines).
Hunk #2 succeeded at 1104 (offset 22 lines).
Hunk #3 succeeded at 1315 (offset 22 lines).

Applying ./patches/010-do-not-add-rt-lib-dirs-when-cross-compiling.patch using plaintext: 
patching file setup.py
Hunk #1 succeeded at 469 (offset 17 lines).
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for --enable-universalsdk... no
checking for --with-universal-archs... 32-bit
checking MACHDEP... linux2
checking EXTRAPLATDIR... 
checking for --without-gcc... no
checking for --with-icc... no
checking for x86_64-linux-gnu-gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for --with-cxx-main=<compiler>... no
checking for x86_64-linux-gnu-g++... no
checking for g++... no
configure: WARNING:
[... 668 lines cut ...]
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/_codecsmodule.c -o Modules/_codecsmodule.o
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/_weakref.c -o Modules/_weakref.o
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/zipimport.c -o Modules/zipimport.o
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/symtablemodule.c -o Modules/symtablemodule.o
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/zlibmodule.c -o Modules/zlibmodule.o
gcc -pthread -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE  -c ./Modules/xxsubtype.c -o Modules/xxsubtype.o
gcc -pthread -c -fno-strict-aliasing -O2 -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes  -I. -IInclude -I./Include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/host/usr/include -I/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/staging_dir/target-powerpc_464fp_musl-1.1.15/host/include  -DPy_BUILD_CORE \
[... cut ...]
make[4]: *** [Modules/getbuildinfo.o] Error 1
make[4]: Leaving directory '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/Python-2.7.13'
Makefile:251: recipe for target '/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/Python-2.7.13/.built' failed
make[3]: *** [/home/dom/Pulpit/Projekty/17.01.0-rc1/WNDR4700/source-68a04db/build_dir/target-powerpc_464fp_musl-1.1.15/host/Python-2.7.13/.built] Error 2
```

hnyman, 2017-01-30:
@build000 
It is the same  SOURCE_DATE_EPOCH thing as in your #3931 like you noticed by yourself

That has nothing to do with the packages themselves... So no need to report it for each package that fails for you.

You might try  -j 1 instead of -j 8.
That might be parallel build issue (although a bit unlikely).

hnyman, 2017-01-30:
@build000 

Just wondering, if this might be related to building from the rc1 tag itself.

Can you show the output of scripts/get_source_date_epoch.sh . It should be something like: 
```
$ scripts/get_source_date_epoch.sh 
1485626731
```

hnyman, 2017-01-30:
@build000 
Great that it worked.
(Note that using version.date is enough, so there is no need to both set the variable and to set time into version.date)

But I am still curious how that fault got launched in the first place. The SOURCE_DATE_EPOCH should be set automatically based on the last git commit during make. So, why did you need to set it?

I have guess, but please confirm it:

* Did you clone & checkout the 17.01.0-rc1 release source via git?
* Or did you download the LEDE 17.01 source as archive (.zip or .tar.gz) instead of cloning/checkouting it with git?  (And then you just expanded the archive to your filesystem and built from it...)

If my guess that you downloaded it as an archive file is right, then you may have actually found a genuine fault in the logic of both Openwrt and LEDE build systems: 

The Makefile logic (in include/toplevel.mk and scripts/get_source_date_epoch.sh) expects to get a value value from git for SOURCE_DATE_EPOCH, but if you are not in a git repo (as you expanded sources from an archive), the SOURCE_DATE_EPOCH is set as empty "". And that then triggers the error in those packages that use the variable without verifying that it is set.
 
I tested myself and both getver.sh and get_source_date_epoch.sh fail to provide a version/time reference:
```
$ wget https://github.com/lede-project/source/archive/v17.01.0-rc1.tar.gz
$ tar -xvf v17.01.0-rc1.tar.gz
$ cd source-17.01.0-rc1/
$ ./scripts/getver.sh 
unknown
$ ./scripts/get_source_date_epoch.sh 
      <-- empty result
```

Ps. the "git way" to get the rc1 sources is:
git fetch --tags
git checkout v17.01.0-rc1

build000, 2017-01-31:
yes - i download tar.gz archive... ;) - no "git clone"

hnyman, 2017-01-31:
> yes - i download tar.gz archive... ;) - no "git clone"

I guessed right, then.

Jow has fixed that in the release tag creation script. Downloading and compiling from .tar.gz should work at the next release (17.01.0 itself or -rc2).

build000, 2017-01-31:
ok.