openwrt/packages #3936: mysql-server: to obsolete for PHP7 and Phpmyadmin

eliz82, 2017-01-31:
https://github.com/openwrt/packages/tree/master/utils/mysql
Maintainter: Jo-Philipp Wich <jo@mein.io>

Any chance to update mysql-server to a newer version? Or having two versions like you have for php? (php5 and php7). Currently the mysql-server available version is 5.1.x

Let me explain the problem I currently have because of this. 
As you know many peoples are switching to the PHP7 because of the big improvement in speed compared with PHP5. Phpmyadmin probably the most used mysql web manager. There is no Phpmyadmin version that support both MySQL 5.1 and PHP7.

Let's have a look at their latest releases:
https://www.phpmyadmin.net/news/2017/1/23/phpmyadmin-466-441510-and-401019-are-released/

Phpmyadmin 4.0.x support PHP 5.2 to 5.6 and MySQL 5.x (including 5.1) 
Phpmyadmin 4.4.x support PHP 5.3.7 to 7.0 and MySQL 5.5
Phpmyadmin 4.6.x support PHP 5.5 to 7.1 and MySQL 5.5 and newer

So at this point we can't use PHP7 and Phpmyadmin because MySQL is to obsolete. We need minimum MySQL 5.5

lucize, 2017-04-01:
can you try this #4221 

regards

micmac1, 2018-05-30:
MariaDB is in master and also in openwrt-18.06 (which will be released in the not too distant future). I suggest this issue gets closed.

closed 2018-08-17 by dibdot