openwrt/packages #3942: feedback needed: openvpn-recipe-<vendor>

stangri, 2017-02-01:
I was suggested to start an issue here to solicit feedback from project leaders.

Due to repeat questions "help me set up a VPN client for specific commercial VPN provider" from users on both OpenWrt and LEDE forums and thanks to the work @cybrnook has done to uci-ize settings for PrivateInternetAccess, I want to make a package which will install required certificates and modify either /etc/config/openvpn or (if luci-app-openvpn is installed) /etc/config/openvpn-recipes to add (in disabled state) openvpn configurations for various PIA servers.

The feedback I'm seeking is:
1. Would a package like that be accepted?
2. Which category does it belong to?
3. What would be a good name? I was thinking openvpn-recipe-privateinternetaccess (in case more openvpn-recipe packages will be added) -- is there a better naming scheme?

Thanks!

stangri, 2017-02-02:
Should this discussion be moved to the mail-list?

yousong, 2017-02-05:
The package will surely become stale and unmaintained as time passes by.  I think a wiki page for these info and caveats will suffice to address the need  for now.

diizzyy, 2019-02-09:
@stangri 
I share yousong's opinion here, I think you can close this issue by now

closed 2019-02-09 by stangri