openwrt/packages #3947: python-setuptools, after updates feed i can not install

banebg, 2017-02-02:
Hello,

I  yesterday updated OpenWRT feeds and installed them. 

After that, when i try to build package Python setuptools i get this:

copying: '/usr/lib/python2.7/doctest.py'
copying: '/usr/lib/python2.7/pydoc.py'
copying: '/usr/lib/python2.7/pydoc_data'
SHELL= flock /home/bane/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/root-brcm2708/tmp-python-pydoc/. /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/root-brcm2708/'
rm -rf /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/root-brcm2708/tmp-python-pydoc
touch /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/root-brcm2708/stamp/.python-pydoc_installed
if [ -f /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/pkginfo/python.default.install.clean ]; then rm -f /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/pkginfo/python.default.install /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/pkginfo/python.default.install.clean; fi; echo "python-pydoc" >> /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/pkginfo/python.default.install
mkdir -p /home/bane/openwrt/bin/brcm2708/packages /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/CONTROL /home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/pkginfo
find /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/ipkg-install -name "*\.pyc" -o -name "*\.pyo" -o -name "*\.exe" | xargs rm -f
install -d -m0755 /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/usr/bin /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/usr/lib/python2.7/site-packages
# Adjust shebang to proper python location on target
sed "1s/.*/#\!\/usr\/bin\/python2.7/" -i /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/install-setuptools/bin/*
sed: can't read /home/bane/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/Python-2.7.13/install-setuptools/bin/*: No such file or directory
make[3]: *** [/home/bane/openwrt/bin/brcm2708/packages/packages/python-setuptools_2.7.13-4_brcm2708.ipk] Error 2
make[3]: Leaving directory `/home/bane/openwrt/feeds/packages/lang/python'
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory `/home/bane/openwrt'
make[1]: *** [/home/bane/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.15_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/bane/openwrt'
make: *** [world] Error 2

I tried to clean,dirclean and distclean and try to build again, but i get this again.

How i can install setuptools?

That is bug on OpenWRT?

Thank you ver much !

commodo, 2017-02-21:
@banebg 
PR https://github.com/openwrt/packages/pull/3970 was merged
can you retry [and close this if resolved] ?

thanks
Alex

cocobear, 2017-04-12:
I got the same error when I used latest version .
The info of branch I used is:

```bash
cocobear@www:~/cc_musl$ git show
commit 871372c42a3fc9c4b33f5c6011742d610a2e5600
Merge: db2ec97 df384cf
Author: Zoltan Herpai <wigyori@uid0.hu>
Date:   Sun Feb 5 13:06:41 2017 +0100

    Merge pull request #373 from wigyori/1043v4v2
    
    TP-Link 1043NDv4 support

cocobear@www:~/cc_musl$ git branch
* chaos_calmer
  master
cocobear@www:~/cc_musl$ 
```

commodo, 2017-04-12:
@cocobear
is the error reproduced on a clean checkout ?

cocobear, 2017-04-14:
@commodo Sorry for the late response, I did make clean then make again, it worked.

commodo, 2017-05-26:
@banebg @cocobear 
can we close this ? 
or is there anything else ?

banebg, 2017-05-26:
Yea,

Thank you very much, all works perfect !

Best regards

closed 2017-05-26 by hnyman