openwrt/packages #3948: freeradius2: compile error

obsy, 2017-02-02:
Maintainer: @dangowrt 
Environment: ar71xx, LEDE 17.01-SNAPSHOT,  LEDE master

Description:
Something wrong goes with compiling freeradius2 (see also buildboot: http://downloads.lede-project.org/snapshots/faillogs/mipsel_24kc/packages/freeradius2/compile.txt ), repeatable on sources lede 17.01-snapshot
```
OpenWrt-libtool: link: ccache gcc -shared  -fPIC -DPIC  .libs/debug.o .libs/dict.o .libs/filters.o .libs/hash.o .libs/hmac.o .libs/hmacsha1.o .libs/isaac.o .libs/log.o .libs/misc.o .libs/missing.o .libs/md4.o .libs/md5.o .libs/print.o .libs/radius.o .libs/rbtree.o .libs/sha1.o .libs/snprintf.o .libs/strlcat.o .libs/strlcpy.o .libs/token.o .libs/udpfromto.o .libs/valuepair.o .libs/fifo.o .libs/packet.o .libs/event.o .libs/getaddrinfo.o .libs/vqp.o .libs/heap.o .libs/dhcp.o .libs/base64.o   -L/var/lib/bbmnt/buildbot/slaves/dave-builder/mipsel_24kc/build/sdk/staging_dir/target-mipsel_24kc_musl-1.1.16/usr/lib -L/var/lib/bbmnt/buildbot/slaves/dave-builder/mipsel_24kc/build/sdk/staging_dir/target-mipsel_24kc_musl-1.1.16/lib -L/var/lib/bbmnt/buildbot/slaves/dave-builder/mipsel_24kc/build/sdk/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/usr/lib -L/var/lib/bbmnt/buildbot/slaves/dave-builder/mipsel_24kc/build/sdk/staging_dir/toolchain-mipsel_24kc_gcc-5.4.0_musl-1.1.16/lib    -Wl,-soname -Wl,libfreeradius-radius-020209.so -o .libs/libfreeradius-radius-020209.so
/usr/bin/ld: .libs/debug.o: Relocations in generic ELF (EM: 8)
/usr/bin/ld: .libs/debug.o: Relocations in generic ELF (EM: 8)
.libs/debug.o: error adding symbols: File in wrong format
collect2: error: ld returned 1 exit status
Makefile:44: recipe for target 'libfreeradius-radius.la' failed
```

c-MM, 2017-02-18:
Yes, i have run into the same problem.

andysyam, 2017-09-14:
why compile build lede many eroor klu like this I have been lazy build openwrt / lede, more errors than the success I can when build

wynter-wang, 2017-11-30:
uuuh...   I have the same just now. But someone have some ideas to fix it?

wynter-wang, 2017-12-01:
OK, I fix it.
1. modify the Makefile:
~/openwrt-homepath/package/feeds/packages/freeradius2/Makefile

- add a line code at the begining of the file
`
    PKG_NAME:=freeradius2
    PKG_VERSION:=2.2.9
    PKG_RELEASE:=1
    PKG_BUILD_DEPENDS:=libtool     # add the line 
`

- then
`
    define Build/Compile
	$(MAKE) -C $(PKG_BUILD_DIR) \
		R="$(PKG_INSTALL_DIR)" \
		INSTALLSTRIP="" \
		all certs install
    endef
    
    change to:

    define Build/Compile
	$(MAKE) -C $(PKG_BUILD_DIR) \
		R="$(PKG_INSTALL_DIR)" \
		INSTALLSTRIP="" LIBTOOL:="$(STAGING_DIR)/host/libltdl/bin/libtool" \
		all certs install
    endef
    ```
`

2. now, compile it.

closed 2018-08-17 by dibdot